trucki-eu/Trucki2Shelly-Gateway #3: ZEPC State über mqtt setzen

jcfuchs, 2023-04-27:
Hallo Trucki,
Wenn man eine Konstanteinspeisung benötigt, um z.B. die Batterien zu schonen/laden, kann man ja über MQTT ACSETPOINTOVR setzen. Dadurch wird ZEPC State auf disabled gesetzt. Um ZEPC wieder zu aktivieren beschreibst du drei Möglichkeiten in der Readme: Reboot, Aktivieren im Webinterface und METEROVR. Um die Aktivierung jedoch zu automatisieren wäre es cool, Senn man diese boolsche Variable über MQTT direct wieder auf true setzen könnte.

Besten Dank
Jan

trucki-eu, 2023-04-27:
Hallo Jan,
Du kannst auf MeterOVR z.b. eine 1 schreiben. Dadurch wird der ZEPC enabled. Das im Trucki2Shelly-Gateway definierte Meter (z.B.b.Shelly) wird die von Dir geschriebene 1 500ms später mit dem ShellyWert überschreiben.

D.h. die Funktion ist heute bereits vorhanden. ZEPC disable = ACSetpointOVR=0, ZEPC enable = MeterOVR=1

Gruß Trucki

jcfuchs, 2023-04-27:
Hallo Trucki,
danke für die schnelle Antwort. Klasse. Habe noch zwei Ideen. Die schreibe ich gerade in einem weiteren Feature Request.
Besten Dank
Jan

closed 2023-04-27 by jcfuchs