trucki-eu/Trucki2Shelly-Gateway #4: Feature Request: Max Power über MQTT setzen

jcfuchs, 2023-04-27:
Hallo Trucks,
ich komme heute mit drei Ideen. Da sie unabhängig sind, erstelle ich drei Feature Requests.

Ich möchte eine Art Ladepriorisierung für die Batterien realisieren. Das soll bedeuten, dass verhindert werden soll, dass Energie aus den Batterien entnommen wird wenn z.B. eine Wolke im Weg ist oder die Last gerade größer als die Sonnenenergie ist. Geladen sollen die Batterien jedoch werden, sofern mehr Sonne als Last zur Verfügung steht.

Um das zu realisieren würde ich gerne meine aktuelle PV-Leistung per MQTT an maxpower schicken. Dadurch nimmt sich der Wechselrichter nur, was die Sonne bietet, die Batterien bleiben unbelastet und ich bräuchte nicht auf deine tolle Regelung zur Nulleinspeisung verzichten.

Grüße
Jan

trucki-eu, 2023-04-27:
Ja macPower über MQTT schreiben kommt mit Version T2SG1.09 in wenigen Wochen.

jcfuchs, 2023-04-27:
Super! :-)

closed 2023-04-27 by jcfuchs