trucki-eu/Trucki2Shelly-Gateway #5: Feature Request: VBAT min max

jcfuchs, 2023-04-27:
Hallo Trucki,
Und wieder MQTT.
Wäre es möglich die Reboot Voltage und Bat Limit Power in deinem Programm zu implementieren. Ich finde es unpraktisch, dass diese wichtigen Werte nur am Wechselrichter direkt eingegeben werden können. Ich möchte diese Werte lieber etwas dynamisch halten. Reboot Voltage im Sommer eher niedriger, weil da die Batterien über den Tag eh voll werden, im Winter dann höher.
Bat Limit wäre variabel praktisch, da die Entladeschlussspannung abhängig von der aktuellen Last einstellbar ist. Gerade bei ollen AGM Batterien macht das Sinn. 
Die Einstellung dafür am Wechselrichter selber wäre dann nur noch eine letzte Sicherheitsabschaltung.

Nun bin ich durch mit meinen Wünschen. :-)

trucki-eu, 2023-04-27:
Ja kommt auch mit der T2SG1.09 . Aber nicht über MQTT. Vielleicht als http. Musst Du dann ausprobieren.

jcfuchs, 2023-04-27:
OK. matt wäre natürlich das beste. aber ich glaube mit dem passenden link / post kann nodered auch damit umgehen.
Danke. Echt Klasse Arbeit was du machst!

closed 2023-04-27 by jcfuchs