trucki-eu/Trucki2Shelly-Gateway #6: feature request: add support for tibber pulse

christophsing, 2023-05-03:
It would be good to have a native support for tibber pulse here. This is helpful for people where a shelly is not possible or not allowed.

trucki-eu, 2023-05-04:
Kannst Du mir mehr Input geben? z.B. Kann man mit CURL den aktuellen Verbrauch auslesen? Braucht man dafür https und einen speziellen Auth Header / API Key?

wauzzi, 2023-05-07:
Die Tibber Json Werte bekommt man Problemlos in den IO Broker mittels NODE RED und können problemlos über den SETPOINTOVR an den SUN XY mittels Script übergeben werden. Klappt bei mir Problemlos.

christophsing, 2023-05-08:
@wauzzi von wo werden die Daten abgerufen? Von der Tibber public API? Diese hat ein rate limit von 100 request pro Minute.

wauzzi, 2023-05-08:
> @wauzzi von wo werden die Daten abgerufen? Von der Tibber public API? Diese hat ein rate limit von 100 request pro Minute.

Ich bin grad unterwegs, daher kann ich nicht direkt schauen. Die Live Daten werden über eine WebSocket Schnittstelle abgerufen

wauzzi, 2023-05-08:
> @wauzzi von wo werden die Daten abgerufen? Von der Tibber public API? Diese hat ein rate limit von 100 request pro Minute.

https://developer.tibber.com/docs/guides/calling-api

christophsing, 2023-05-08:
@wauzzi 
es sind sogar nur 100 requests pro 5 Minuten:

Rate limiting
There is a rate limit of 100 requests in 5 minutes per IP address intended to protect the API. Note that prices are calculated once per day in the afternoon (for Norway and Sweden they are first preliminary and later finalized potentially with minor changes after currency rates are confirmed). You can use priceInfo.today and priceInfo.tomorrow for fetching them up front instead of only using priceInfo.current for the current hour.

For more information see [Calling API](https://developer.tibber.com/docs/guides/calling-api)

wauzzi, 2023-05-08:
> @wauzzi es sind sogar nur 100 requests pro 5 Minuten:
>
> Rate limiting There is a rate limit of 100 requests in 5 minutes per IP address intended to protect the API. Note that prices are calculated [...]

Bei einer WebSocket Verbindung handelt es sich nach Aussage meines Arbeitskollegen (Entwickler) nicht direkt um einen request, sondern eine Art Live Stream.

Die Request wo du meist, beziehen sich glaube ich nur auf den Abruf der aktuellen Preise.

marcokiewit, 2023-06-10:
> aber gerne doch - es gibt eine Anleitung wie das klappen könnte - erscheint mir jedoch nicht besonders praktikabel. https://github.com/MSkjel/LocalPulse2Tibber
> 
> eine direkte Anbindung an die Tibber APi ist möglich aber aufgrund des Rate limits nicht sinnvoll: https://developer.tibber.com/docs/overview
> 
> Ich habe eine Anfrage an Tibber gesendet und melde mich nochmal mit Infos.

Hast du auf deine Anfrage schon eine Antwort bekommen?

christophsing, 2023-10-24:
https://github.com/marq24/ha-tibber-pulse-local

Tekklab, 2023-10-25:
> https://github.com/marq24/ha-tibber-pulse-local

Danke fürs teilen, das funktioniert tatsächlich hervorragend nach Freischaltung der Pulse. Wenn ich das richtig gelesen habe, könnte man nun mal versuchen die Daten über http://admin:{pwd}@{host}/data.json?node_id=1 zu nutzen

Tekklab, 2023-11-15:
Heute sind Trucki Platine und Meanwell Stick gekommen, Tibber Vertrag startet ab Januar, Zugriff etc. funktionieren aber, ich werde also mit der Tibber Integration starten. Frage in die Runde: Gibt es da grundsätzlich Interesse an einer "echten" Tibber Pulse Anbindung oder reicht auch eine Anbindung über HA, das wäre nämlich relativ easy und funktioniert bereits jetzt:

1. die Schritte auf https://github.com/marq24/ha-tibber-pulse-local befolgen, d.h.
a) Tibber Pulse Frontend dauerhaft öffnen (ist recht einfach und gut beschrieben)
b) HACS Integration in HA durchführen und die lokalen Werte sind in HA zu finden

Läuft bei mir seit 3 Wochen sehr stabil, sehr viel stabiler als die Tibber API, die ich zusätzlich in HA angebunden habe und die im Zeitraum einige Aussetzer hatte.

2. Jetzt kann man sich einfach den Wert des im Schritt 1 erzeugten Sensors als JSON direkt von der lokalen HA API holen:
`curl -H "Authorization: Bearer TOKEN" -H "Content-Type: application/json" http://HA-IP:8123/api/states/sensor.PULSE-SENSOR` und im Trucki2Shelly hinterlegen. Das JSON Objekt sieht dann so aus:
`{"entity_id":"sensor.pulse_0100100700ff","state":"446.85","attributes":{"state_class":"measurement","unit_of_measurement":"W","device_class":"energy","icon":"mdi:meter-electric","friendly_name":"Tibber Pulse Bridge local polling Wirkleistung aktuell"},"last_changed":"2023-11-15T17:53:49.487536+00:00","last_updated":"2023-11-15T17:53:49.487536+00:00","context":{"id":"XXX","parent_id":null,"user_id":null}}%`

Das ist dann noch keine echte Tibber Integration, sondern eine Tibber Pulse Integration, interessant wird's dann beim Laden des Akkus zu günstigen Zeiten über das Trucki2Meanwell Gateway. Hier schaue ich mir gerade die von der Tibber API zur Verfügung gestellten ["PriceLevel"](https://developer.tibber.com/docs/reference) an, ein Ansatz wäre: Laden des Akku bei "VERY_CHEAP" und "CHEAP" und die Nutzung des SUN bei "EXPENSIVE" und "VERY_EXPENSIVE" Strom.

Gibts Interesse und/oder bereits gesammelte Tibber Erfahrung?

Ultrahoschie, 2024-01-05:
Ich hätte definitiv Interesse, da damit das System rund wird! Zudem könnte man von günstigen Preisen bei eventuellen zuladen und endladen profitieren.

Ultrahoschie, 2024-01-06:
Zudem ist es so, das alle drei Phasen ausgelesen werden, ähnlich dem Shelley!