trucki-eu/Trucki2Shelly-Gateway #7: Max Power Set bei Reboot auf NULL

wauzzi, 2023-05-07:
Sobald einer der beiden SUN2000 mit Trucki Platine Stromlos gemacht wurde, fliegen die Max Power Set Werte wieder auf Null und müssen erneut gesetzt werden. Bitte um Fix

trucki-eu, 2023-05-08:
- Mit welcher Version?
- Wurde MaxPower über das WebInterface oder HTTP/MQTT geschrieben?
- Save gedrückt nach dem Schreiben im WebInterface?
- Steht der Wert in der log.txt nach dem Reboot richtig drinn?
- Wurde ein FactoryReset/ Wipe schon probiert?

wauzzi, 2023-05-08:
Ich habe glaub das Problem gefunden. Als Default wird über Mqtt der Wert Null geschrieben. Setzt man nun den max Power Wert über das WebIF auf den gewünschten Wert und startet den Sun neu, wird der Wert Null aus dem Iobroker Null wieder übernommen. Es wäre evtl gut, wenn nach einer Änderung im WebIF die Parameter auch über Mqtt an den Iobroker übertragen werden.

Mit welcher Version?

Aktuellste
Wurde MaxPower über das WebInterface oder 
WEBif

HTTP/MQTT geschrieben?
Http
Save gedrückt nach dem Schreiben im WebInterface?
Ja
Steht der Wert in der log.txt nach dem Reboot richtig drinn?
Muss ich noch schauen 
Wurde ein FactoryReset/ Wipe schon probiert?
Ja

closed 2023-08-14 by trucki-eu