trucki-eu/Trucki2Shelly-Gateway #8: Warum ist das kein OpenSource projekt?

delacor, 2023-05-28:
Moin, mich würde interessieren wieso es hier keinen code zu gibt sondern nur Binarys. Das finde ich sehr schade.

Liebe Grüße

tcurdt, 2023-07-28:
Da GNU licensed code verwendet wird ist das auch nicht wirklich optional.

[image]

trucki-eu, 2023-07-28:
Ja, dass hat man davon, wenn man transparent ist... 

Ich habe den Code nicht mehr hochgeladen, da gute gemeinte Verbesserungsvorschläge und Fragen, 100% der verfügbaren Zeit in Anspruch genommen haben.

Was ratet ihr mir? Dank des ESP Webservers ist es leicht die GNU Bibliotheken wieder rauszuwerfen?

tcurdt, 2023-07-28:
Ist ein tolles Projekt und deswegen etwas Gutes kaputt machen wäre schade.
Zumal das Kind jetzt schon in den Brunnen gefallen ist.
Den Source Code der aktuelle Version bist Du eigentlich schuldig.

Und auch MIT/BSD erfordern zumindest den Disclosure.

Das ist schon gut wie Du es gemacht hast. 

Wenn es nur um das Feedback geht würde auch vorschlagen den Code zu veröffentlichen und Dich von den Verbesserungsvorschlägen und Fragen einfach ein bisschen besser abzuschirmen. Muss man vielleicht ein bisschen üben - aber das geht :)

Nur weil es Open Source ist entsteht daraus ja nicht gleich eine Verpflichtung, dass Du die Sachen auch umsetzen musst.
Einmal im Monat PRs anschauen sollte ja vielleicht machbar sein - falls es welche gibt.

Mach vielleicht die Discussions im Github Projekt an und lass die Leute sich untereinander helfen.
Und schau halt nur rein wenn Lust und Zeit hast.

Bei den Ticket die reinkommen gibt es zum Teil Hilfe durch Bots (not a fan) oder Du achtest nur darauf, dass sie Dinge klar eingeordnet sind. Zum Beispiel Bug/Enhancement/NeedsVolunteers.

Nur so meine 2 cents nach vielen Jahren Open Source.

Die Frage ist halt wo Du mit dem Projekt hin möchtest.

trucki-eu, 2023-07-28:
Vielen Dank für deine (liebevolle) Beratung. Ja keine Ahnung wo das ganze hinläuft. Vielleicht wird es auch Zeit Github langsam zu verlassen und eine eigene Seite für das Projekt aufzusetzen.

Grundsätzlich kann man sich bei mir melden und nach dem Code fragen. Ich glaube die Verteilung als physisches Medium (z.b. CD ) ist ja auch vorgesehen?

tcurdt, 2023-07-28:
> Vielleicht wird es auch Zeit Github langsam zu verlassen und eine eigene Seite für das Projekt aufzusetzen.

Ich hoffe das eine schließt das andere nicht aus.

> Grundsätzlich kann man sich bei mir melden und nach dem Code fragen. 

Fände ich persönlich recht schade wenn es so läuft.

> Ich glaube die Verteilung als physisches Medium (z.b. CD ) ist ja auch vorgesehen?

Wie jetzt? :)

littleyoda, 2023-08-20:
> Ja, dass hat man davon, wenn man transparent ist...
> 
> Ich habe den Code nicht mehr hochgeladen, da gute gemeinte Verbesserungsvorschläge und Fragen, 100% der verfügbaren Zeit in Anspruch genommen haben.

Mein Vorschlag:
- Packe den Sourcecode auf Github
Aus dem Thema kommst du wegen der Lizenzthematik nicht raus. Die Sache mit dem Versand von CDs willst du nicht. Macht ja nur unnötige Arbeit.
- Deaktiviere die Issue hier
Damit kommen die User gar nicht erst in Versuchungen nach Feature X zu fragen. Deine E-Mails zu suchen macht deutlich mehr Arbeit, als hier ein Issue zu öffnen.
- Oder mache dich von den Anforderung von anderen frei.
Schreibe im Zweifel einfach, dass Feature X nicht geplant ist du dich aber über ein PR von jemanden anderen freust. Damit ist dann klar, dass du es nicht implementieren wirst.
- Oder schreibe, dass du es als Feature-Complete ansiehst und gar keine neuen Features aufnimmst.

TL;DR Kommuniziere deine Erwartungshaltung und passe Github entsprechend an.

erxbout, 2023-09-05:
Würde mich auch über source freuen

Falls Interesse besteht im Github zu managen gibts sicher jemand der helfen wollen würde (hallo hier ich * wink * ;))
Grundsätzlich kann und muss jeder hier verstehen, dass dieses Projekt Freizeit ist und man halt nur ab und an was machen kann und will..

Wenn man das klar irgendwo hinschreibt und die leute PRs stellen lässt müsste man ja (so wie ich verstanden habe) die Richtung des Projektes in den Issues lenken können und über PRs code akzeptieren der sinnvoll ist..

Und wenns irgendwem nicht passt oder nicht schnell genug geht kann er immer einen Fork machen und so sein Projekt verfolgen

DocSnyd3r, 2023-10-30:
Hallo @trucki-eu wie kann man sich denn melden und nach dem Code fragen?
Ich habe hier alles mögliche abgesucht und versuche das Layout sowie die sourcen für das Trucki2MeanWell Gateway zu finden.
Sprich um das MeanWell Ladegerät ansteuern zu können, ich lande aber immer wieder beim Wechselrichter, der ist für mich aber erst später interessant. Ich möchte mir das aktuell so bauen dem Ladegerät einfach mittels MQTT sagen zu können mit wieviel Energie es gerade laden darf.

0xFEEDC0DE64, 2024-08-13:
was passiert, wenn jemand per email die sourcen beokmmt und dann unabsichtlich bei github leicht ankommt und öffentlich eincheckt?

mi-hol, 2024-11-10:
@trucki-eu sehr schade, dass ein so gutes Projekt closed source ist. Ich kann die Argumente "Ich habe den Code nicht mehr hochgeladen, da gute gemeinte Verbesserungsvorschläge und Fragen, 100% der verfügbaren Zeit in Anspruch genommen haben."
nachvollziehen, aber es gibt bessere Möglichkeiten dieses Problem in den Griff zu bekommen, als zu "closed source" zu wechseln.
Es erfordert Vertrauen beim "Vater" der Software, dass man mit einem Team von Leuten (mit anderen Skills), als Team erfolgreicher sein kann als nur eine einzelne Person!
Angebote die Last auf weitere Schultern zu verteilen wurden gemacht.

Ich weiss leider durch eigene Erfahrung, wie schnell die eigenen Fähigkeiten, durch ein unvorhergesehenes Ereignis zunichte gemacht werden können. Bei einem Projekt, das durch nur einen Einzelnen vorangetrieben wird, ist dann Schluß!
Das kann jeden plötzlich treffen, niemand ist unkaputtbar oder unsterblich.

Ich  hatte überlegt mir ein "Trucki" System zu bauen, bin aber wegen obigem Risiko wieder davon abgekommen.

Vielleicht gibt es ja doch noch eine Chance?

0xFEEDC0DE64, 2024-11-12:
ich bin für eine copyright-verletzungs-meldung bei github mit anschließender takedown notice, man kann nämlich aus den binaries raus lesen, dass bis heute immer noch GPL libraries verwendet werden und dies eine Verletzung der Lizenz darstellt.
Habe gehört die bei github reagieren immer relativ schnell, wenn man klar beweisen kann, dass die Verletzung vorliegt.
Eventuell auch me-no-dev und co ins Boot holen und sie mal fragen, was sie davon halten.
ich mach mich gleich ans Schreiben...

tcurdt, 2024-11-12:
@0xFEEDC0DE64 Ist die Frage was das Ziel ist. Dann ist einfach nur das Repo hier weg. Macht das die Sache besser?

tcurdt, 2024-11-12:
Toll wäre ein Dialog. Zur Not auch über die ganzen Youtuber die das Projekt pushen.

theunknownstarman, 2024-11-12:
> @0xFEEDC0DE64 Ist die Frage was das Ziel ist. Dann ist einfach nur das Repo hier weg. Macht das die Sache besser?

nein.

0xFEEDC0DE64, 2024-11-20:
macht die sache schon besser, weils dann diese dumme closed source lösung nicht mehr gibt und vielleicht jemand anderer (gescheiterer?) eine ordentliche Lösung entwickelt und die auch mit der Welt teilt.
Aktuell ist halt einfach der need nach einem rewrite zu gering, weil es diese dumme closed source lösung immer noch gibt.
Bin gespannt, wann mein report bei GitHub bearbeitet wird :P