trucki-eu/Trucki2Shelly-Gateway #10: Stick arbeitet nach Neustart nicht

66xerxeS66, 2023-06-15:
Hi,

ich habe den Lumentree 600W Wechselrichter und benutze diesen zur reinen Nachteinspeisung. Nachdem die Zeitfunktion im WR nicht arbeitet habe ich das ganze mit einem Shelly Plug S gelöst, welcher die AC Seite 2 h vor Sonnenuntergang ein und 2 h nach Sonnenaufgang ausschaltet.
Meistens klappt es ganz gut mit dem WR. Dieser startet nachdem die AC Seite an ist und der Stick fängt an zu regeln. Ab und zu jedoch startet der WR und der Stick ist zwar an, der WR speist jedoch das Maximum ein, was ich im WR eingestellt hab ein. Den Stick muss ich dann rebooten, damit er mit der Regelung beginnt. 
MQTT ist deaktiviert, da ich kein HomeAssistant hab. 
Ist das Problem bereits bekannt? 
Wird es in Zukunft eine Funktion zur Nutzung der Zeitschaltuhr geben? 

Schöne Grüße

P.S

Danke für deine super Arbeit und deine Bemühungen

[image]

Geniuszeu, 2023-06-15:
Hallo zusammen,genau dasselbe Problem habe ich mit dem Lumentree 600W. Gruß Eugen

trucki-eu, 2023-06-16:
Guten Morgen,

vielen Dank für Euer Feedback. Ich habe gerade die Version T2SG V1.11 unter Releases hochgeladen:

https://github.com/trucki-eu/Trucki2Shelly-Gateway/releases

Die V1.11 sendet beim Start 3x 0W an den SUN Wechselrichter, damit dieser ganz sicher vom Mppt Modus in den "Stick-Steuer" Modus wechselt.

Ja, ich arbeite an einer zeitgesteuerten Nachtfunktion für das T2SG.

Viele Grüße,
Trucki

MadeMartin, 2023-06-17:
Moin,
ich hatte die gleichen Probleme wie 66xerxeS66. Mit v1.11 läuft die 1000er Lumentree-Anlage wieder fehlerfrei.
VG
Martin

66xerxeS66, 2023-06-18:
Hi Trucki,

Update war super! Läuft seitdem perfekt!

Gruß

xerxeS

Geniuszeu, 2023-06-18:
Hallo zusammen,bei mir läuft auch super.  Prima Arbeit Danke --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

closed 2023-08-14 by trucki-eu