trucki-eu/Trucki2Shelly-Gateway #12: Nach Neustart von Home Assistant -> keine MQTT Verbindung

BayernJoJo, 2023-07-08:
Hallo, wenn ich mein Home Assistant starte, weil ich was in der YAML verändert habe, oder nach einer neuen Integration, ist die MQTT Verbindung weg.
Ich muß dann den Lumentree 600 ausschalten oder den Stick abziehen und wieder draufstecken, damit er die MQTT Verbindung wieder aufbaut.
Ist das Problem bekannt?

Grüße

trucki-eu, 2023-07-09:
Hallo BayernJojo,
mit der aktuellen Version V1.11 sind keine MQTT reconnect Probleme bekannt
Gruß,
Trucki

BayernJoJo, 2023-07-09:
Hallo,
Ich hab die V1.11 drauf, das Problem besteht weiterhin.
Immer nach einem Neustart von HA, bleiben die MQTT Werte des Wechselrichters auf „unavailible“ 🤷🏻‍♂️

Gruß
JoJo

trucki-eu, 2023-07-09:
Hallo Jojo,
wie geschrieben ist aktuell kein Problem in der Richtung bekannt. Ich nutze selbst HA und reboote jede Nacht.
Zur Lösung kannst Du das Log des MQTT Brokers prüfen und die Kommunikation im Fehlerfall mit MQTT Explorer beobachten und simulieren. Eventuell liegt es auch an der grundsätzlichen Wlan-Verbindung.
Viel Erfolg,
Gruß,
Trucki

BayernJoJo, 2023-07-13:
Servus,

hab leider immer noch keine Lösung für mein Problem gefunden. Was mir aber aufgefallen ist, der Mosquito MQTT Broker zeigt beim Stick immer noch die Firmware 1.09 an. Obwohl auf dem Stick schon 1.11 drauf ist.
Werde noch paar Sachen probieren, vorerst bleibt mir nur immer den Stick aus und wieder an zu stecken.

BayernJoJo, 2023-07-18:
Hallo nochmal,
zu meinem Problem hab ich folgenden Log:
Error 'required key not provided @ data['command_topic']' when processing MQTT discovery message topic: 'homeassistant/number/T2SGEA4D4F_MAXPOWEROVR/config', message: '{'name': 'T2SGEA4D4F Set max. power', 'command_topic:': 'T2SGEA4D4F/MAXPOWEROVR', 'device_class': 'power', 'unit_of_measurement': 'W', 'unique_id': 'T2SGEA4D4F_MAXPOWEROVR', 'device': {'identifiers': 'T2SGEA4D4F', 'manufacturer': 'Trucki', 'model': 'T2SG', 'name': 'T2SGEA4D4F', 'sw_version': '1.09'}, 'platform': 'mqtt'}'

Kann mir da jemand weiterhelfen?
Gruß
JoJo

trucki-eu, 2023-07-18:
Hast Du mal versucht des HADiscoveryMode im T2SG zu deaktivieren und die MQTT Werte manuell in die HA config einzutragen.
So in der Art?
https://github.com/trucki-eu/Trucki2Shelly-Gateway#homeassistant-mqtt-readwrite

BayernJoJo, 2023-07-19:
Danke,
manuell funktioniert es.  :-)
Aber warum, weiß ich nicht....
Kann mir das einer erklären? Es muß ja dann an der "HomeAssistantDiscovery" liegen, er zeigt mir im MQTT Broker auch immer noch Version 1.09 an, obwohl ich schon auf 1.11 bin.

Naja, Hauptsache es funktioniert jetzt. 
Vielen Dank

ditec-de, 2023-08-06:
Ich vermute, 
dass dein Mosquitto Broker die falsche Version in seiner Datenbank in den "Retained Messages" gespeichert hat.

Eine mögliche Lösung wäre: 
1.) Mosquitto stoppen 
2.) Mosquitto Datenbank löschen 
3.) Mosquitto starten

Eine andere Möglichkeit (gezieltes Löschen der einzelnen Message):
mosquitto_pub -h hostname -t the/topic -u username -P password -n -r -d

Oder das Mosquitto HA Plug-In deinstallieren und dann neu installieren würde auch gehen...

Viel Erfolg!

closed 2023-08-14 by trucki-eu