trucki-eu/Trucki2Shelly-Gateway #13: Verbindungsabbrüche zwischen MQTT und Homeassistant

Eismann, 2023-07-18:
Hallo,

ich habe gestern den Stick (an einem Lumentree 600) in HA eingebunden. Dabei wird die MQTT Verbindung  hergestellt aber nach 3-4 Minuten getrennt. Nach weiteren 3-4 Minuten wird die Verbindung neu aufgebaut und das dann immer im Wechsel.
Ist das beabsichtigt oder ein Fehler? So sind die Sensoren in HA für ein Echtzeitmonitoring relativ nutzlos.

Grüße

trucki-eu, 2023-07-18:
Hallo Eismann,
mir ist derzeit kein Fehler bekannt. Im Stick gibt es keinen 3-4 Minuten Timer. 
Mit dem Mqtt Explorer kannst Du sehen ob der Stick dann noch Daten sendet.
Könntest Du bitte noch ein paar mehr Informationen geben:
z.B.
- Log des Brokers
- Ist der Stick noch erreichbar? z.B. Ping
- Ist der Stick noch per MQTT Explorer erreichbar?
Viele Grüße,
Trucki

Eismann, 2023-07-19:
Guten Morgen,

Du bist aber schnell. 

Der Fehler ist behoben. Das ganze Fehlerbild erschien mir ohnehin etwas komisch.

Es war kein Problem des Sticks oder des MQTT Brockers sondern ein IP Adresskonflikt. 
Irgendwann muss ich bei einer Bastelei mal temporär den DHCP Range verstellt haben. Daher hatte dort ein Gerät eine IP bekommen, die normal in einem Range für feste IP´s liegt. Diese IP hatte nun auch der Stick von mir bekommen und nun gab es natürlich einen Konflikt, Mir ist das nicht sofort aufgefallen,, da der Stick ja Anfangs lief
Nachdem das bereinigt ist, läuft bisher Alles stabil.

Gruß

closed 2023-07-19 by Eismann