trucki-eu/Trucki2Shelly-Gateway #14: Kein AC Setpoint im Trucki2Shelly Gateway

myjobismyhobby, 2023-07-25:
Ich habe den Trucki2Shelly Gateway Stick, zusammen mit einem Lumentree SUN 2000 eingerichtet. Das Webinterface funktioniert soweit, ich sehe Batteriespannung, Netzspannung, etc. Auch die Werte vom Shelly kommen korrekt rüber und melden an den ZEPC, dass eingespeist werden soll.

Am WR bleibt der AC Setpoint aber immer bei "0" stehen. D.h. es erfolgt keine Regelung über den Stick. Die geregelte Einspeisung funktioniert daher bei mir aktuell nicht. Was mache ich falsch? Gibt es noch eine Anleitung für die Einstellungen am WR? Ich habe die Einstellungen nach Vorgabe von "Der Kanal" im WR getätigt (Haken bei "Bat or solar limited Power Mode gesetzt" und "Internal Limiter" auf "Off").

trucki-eu, 2023-07-25:
Hallo,

Ist das vielleicht ein Lumentree von Alibaba oder Ebay? Der Stick braucht eine spezielle Firmware auf dem Lumentree. Nur die Lumentree's aus dem Amazonshop von Ask4It haben diese Firmware. 

Falls das der Grund ist, könntest Du alternativ meine RS485 Platine verwenden. Sie ist mit alles SUN 0.6/1/2000 kompatibel. Und die Software ist die Gleiche wie auf den Stick.

Viele Grüße Trucki

myjobismyhobby, 2023-07-25:
Hallo Trucki,

ja das ist mir bekannt, nein der Lumentree ist von euch (zumindest wäre mir nichts anderes bekannt).

Das Bootlogo kommt auch korrekt. Das Logo zeigt die Version 6.1

Danke und VG

Tim

Von: trucki-eu
Gesendet: Dienstag, 25. Juli 2023 13:09
An: trucki-eu/Trucki2Shelly-Gateway
Cc: myjobismyhobby; Author
Betreff: Re: [trucki-eu/Trucki2Shelly-Gateway] Kein AC Setpoint imTrucki2Shelly Gateway (Issue #14)

Hallo,
Ist das vielleicht ein Lumentree von Alibaba oder Ebay? Der Stick braucht eine spezielle Firmware auf dem Lumentree. Nur die Lumentree's aus dem Amazonshop von Ask4It haben diese Firmware.
Falls das der Grund ist, könntest Du alternativ meine RS485 Platine verwenden. Sie ist mit alles SUN 0.6/1/2000 kompatibel. Und die Software ist die Gleiche wie auf den Stick.
Viele Grüße Trucki

trucki-eu, 2023-07-25:
Ich verstehe. Funktioniert die manuelle Ansteuerung über den AC Send im SUN1 Menü?
Sind die Limiter im Display vom Inverter deaktiviert?

myjobismyhobby, 2023-07-25:
Lt. Christian muss doch im Invertermenü der Haken bei „Bat or Solar Limited Power Mode“ gesetzt sein, oder?Der Haken bei „Internal Limiter“ ist deaktiviert.Oder müssen alle Haken im Invertermenü raus? Nein, so wie es aussieht, funktioniert auch die manuelle Übermittlung via „AC -> send“ im SUN1 Menü nicht.

myjobismyhobby, 2023-07-25:
Und nochmals kommt mit komisch vor:Die Temperatur wird im Gateway ebenfalls nicht mit angezeigt.In dem Video von Christian aber schon.

trucki-eu, 2023-07-25:
Das sind aber die typischen Merkmale für falsche Firmware auf dem SUN: keine Temperatur und AC Setpoint geht nicht...

myjobismyhobby, 2023-07-25:
Okay, und was mache ich nun? Ich würde gerne deine Regelung mit dem WR nutzen. Aktuell funktioniert es aber nicht.

trucki-eu, 2023-07-25:
Klar, schreib mir doch bitte eine Email an you*ube(ät) tru*ki.eu ... *musst Du durch t und c ersetzen.

closed 2023-08-14 by trucki-eu