trucki-eu/Trucki2Shelly-Gateway #15: Trucki2shelly app`? Neue Optionen?

shake2k, 2023-07-27:
Hi bin neu hier und wollte fragen ob eine App in Planung ist?

Das Webinterface ist schon echt nice, was bis jetzt alles kam nur so aus Interesse ;D

Und dann noch eine Idee die man vielleicht einbauen könnte. Hätte gerne, dass man

einstellen könnte, dass beim Entladen ab 24V bis cutoff ein Maximalwert automatisch

gesetzt wird zB 80 Watt MAX :P finde es sollte dann etwas entspannter bis 23V entladen

werden.  Nur eine Idee vielleicht wünscht sich das hier wer anders auch ;D

trucki-eu, 2023-07-28:
Probier mal den Cutuff im Sun selbst aus. Der macht das so und da dies unerwünscht war habe ich den Cutoff im T2SG programmiert.

shake2k, 2023-07-28:
Aha!!! Das mit der App ist wohl noch geheim verstehe  ;D

Das ist ja interessant mit dem internen Cutoff, dass der vorher dann weniger Leistung macht bis zum Cutoff.

Habe Max Power auf 416W eingestellt, da läuft der Lüfter nur mit ca 22 Watt statt 30Watt wenn Max Power 600w ist.

Der Spannungsmesswert vom SUN im Vergleich zum Victron Batterie Sense hat 0,6V unterschied.

Bedeutet wenn ich den Stick auf 22,3 cutten lasse ist der Victron gerade bei exakt 23V und den Internen

hatte ich bisher auf 22V weniger geht ja nicht ;D einpendeln nach cutoff tut sich der wert nach 2 Std dann bei 23,6V bisher.

Gut dann werde ich am Wochenende mal unter Beaufsichtigung die Werte einfach miteinander tauschen.

Im Sun stelle ich dann mal 22,5 ein und im Trucki Stick dann 22,1 oder so, bin echt gespannt.

Danke für Deinen Tip werde dann berichten wie es lief ;D

shake2k, 2023-07-29:
hat wirklich geklappt aber ich bleibe doch beim Stick wegen den paar sekunden und minuten ist es denke total wurscht ;D