trucki-eu/Trucki2Shelly-Gateway #16: LumenTree 600 mit Trucki Stick regelt nicht

Penny-001, 2023-07-30:
Hallo,

ich habe einen LumenTree 600 mit einem Trucki Stick und versuche das nur die Leistung ( Meter) die gerade im Haus verbraucht werden, auch nur eingespeist wird,

leider ohne Erfolg.

Es wird immer die volle Leistung eingespeist.

Was mache ich falsch?

Ich habe nur Solarpanels angeschlossen und keine Batterie.

[image]

[image]

 

Gruß

trucki-eu, 2023-07-30:
Shelly 3EM richtig herum eingebaut?
Target=30W, Average =10 ?

Penny-001, 2023-07-30:
Ist kein Shelly 3EM, sondern vom IOBROKER ein Datenpunkt.
Es werden mehrere Shelly's addiert und in den Datenpunkt gespeichert.
Der Wert vom dem Datenpunkt wird ja auch in Meter richtig angezeigt.
Target= 30 u. Average =10 ist richtig.

trucki-eu, 2023-07-30:
Schreibst Dürer MQTT an meterovr und versuchst den ZeroExportController des T2SGs zu verwenden?
Dann ist es wichtig, dass dein Inverter hinter dem Meter hängt und die gemessene Leistung des Meters sich um die Leistung des Inverters reduziert.
Wenn das nicht möglich ist, dann musst Du die die Regelung selber bauen und an ACSetpointOVR die Invterter Leistung schreiben.

Penny-001, 2023-07-30:
Unter Meter habe ich diesen Link drin:
http://192.168.1.67:8087/get/0_userdata.0.ShellyPlusPlugS.Power_Liste

Json keys: val

nimmt er auch, und zeigt den Wert ja richtig an.

Über MQTT gebe ich nichts ein.

Mit diesem Befehl geht das bei mir:
curl "http://192.168.1.114/?modbus_reg=0&modbus_value= xxx " > NULL

trucki-eu, 2023-07-30:
Und hängt der Sun auch mit an dem Plug?

Penny-001, 2023-07-30:
nein.

trucki-eu, 2023-07-30:
Deshalb geht der Sun auf Maximum. Das Meter will/muss die Leistung des Suns sehen können. Siehe oben.

Penny-001, 2023-07-30:
Jetzt habe ich es verstanden !

und es läuft!!!!

Vielen Dank für die Hilfe.

Gruß
Björn

trucki-eu, 2023-07-30:
Prima! Wie hast Du es gelöst? Den Sun mit an den Plug angeschlossen? 
Kann der Plug auch negative Leistungen messen? Sonst könnte es zu Problemen kommen, wenn eine große Last abschaltet!

Penny-001, 2023-07-30:
ich habe insgesamt 12 Plug's die ich  addiere und in einem Datenpunkt speicher. Der Sun hat einen eigenen Plug der dann davon abgezogen wir.
Es soll auch nur eine Notlösung sein, bis ich ein Shelly 3 EM im Verteilerkasten habe.

Gruß

closed 2023-08-14 by trucki-eu