trucki-eu/Trucki2Shelly-Gateway #17: Leistung (Meter) Angabe von Tasmota Lesekopf klappt nicht

H1ght0w3r, 2023-08-04:
Moin

Ich habe eine Lumentree 600 mit einem Trucki Stick . Die verbrauchte Leistung (Meter) soll von einem 
Tasmota Lesekopf geholt werden.

￼
[image]

Hier ist das folgende Script hinterlegt:
￼
[image]

Die verwendete URL ist:
http://192.168.178.78/cm?cmnd=status%2010

Als Ergebnis bekomme ich:
{"StatusSNS":{"Time":"1970-06-11T05:35:26","":{"Total_in":2173.5,"Total_out":349.3,"Power_curr":-136.39}}}

Was muss ich als JSON Key eintragen damit ich den Wert von „Power_curr“ angezeigt bekomme?
￼
[image]

Ich habe bereits verschiedene Kombinationen ausprobiert:

- StatusSNS,Total_in,Total_iout,Power_curr
- StatusSNS,Power_curr
- StatusSNS,0,0,Power_curr  
- Power_curr

Alle sind falsch 

Gruß
H1ght0w3r

trucki-eu, 2023-08-05:
Schau Dir mal mein Script für den Json Keys: (StatusSNS,SML,Power_curr) an:

`>D
>B
=>sensor53 r
tper=20
>M 1
+1,13,o,0,300,SML,15,32,2F3F210D0A,063030300D0A
1,1.8.1*00(@1),Power_curr,W,Power_curr,2`

https://github.com/trucki-eu/Trucki2Shelly-Gateway#tasmota-instead-shelly

Bei Dir fehlt das SML. D.h. dein 2. Json Key ist "" .

Gruß,
Trucki

H1ght0w3r, 2023-08-07:
Ich habe mein Script um SML erweitert:
----
>D
>B
=>sensor53 r 
>M 1
+1,3,s,0,9600,SML
1,77070100100700FF@1,Leistung,W,Power_curr,2
1,77070100010800FF@1000,Verbrauch Total,kWh,Total_in,1
1,77070100020800FF@1000, Netzeinspeisung,kWh,Total_out,1
#
---

Darauf hinderte sich die Ausgabe der URL auf:
{"StatusSNS":{"Time":"1970-06-14T09:49:19","SML":{"Power_curr":335.77,"Total_in":2195.0,"Total_out":357.6}}}

und mit dem JSON Key: 
StatusSNS,SML,Power_curr

hat es jetzt geklappt.

Danke Trucki

Gruß
H1ght0w3r

closed 2023-08-14 by trucki-eu