trucki-eu/Trucki2Shelly-Gateway #19: Regelung mit Shelly 3em - Fehlfunktion / Bleib hängen

Jollinar, 2023-08-10:
Seit einigen Tage habe ich das Problem das nach einiger Laufzeit die Regelung nicht mehr korrekt funktioniert der Shelly sendet wie gewohnt die Daten und das auch korrekte aber die Steuerung zeigt folgendes. (Beispiel)

SUN1 (Local) = 122,40 W (AC Setpoint 122,40 W)
Meter = 176,8 W
ZEPC = 600 W ( Target 5 | Average 30)

Eine Regelung erfolgt nicht mehr wirklich zwar bewegen sich AC Setpoint und Meter aber ZEPC bleibt bei 600 W stehen.
Ein Reboot des Sticks hilft hier leider nicht, nur ausschalten der Wechselrichters behebt das Problem, aber nicht dauerhaft nach einer gewissen Laufzeit kommt das Problem wieder.

Ist das Problem mit der 1.12 bekannt kann ich auf die 1.11 zurück ?

Software Version 1.12 
+ Lumentree 600 
+ Trucki Platine 
+ TruckiStick 
+ Shelly 3 EM 
+ 7,8KW 48V Akku
+1,5 KW Peak (4x380Watt)

trucki-eu, 2023-08-11:
Hallo Jollinar,
nein, in der Richtung ist nichts bekannt. Wie groß ist das PowerLimit im Fehlerfall? Kannst Du einige Screenshots posten?
Gruß,
Trucki

Jollinar, 2023-08-14:
Hi Trucki, 

das Powerlimit ist bei 600W eingestellt, leider ist es bisher noch nicht wieder aufgetreten sobald ich den Fehler aber wieder haben bekommst du eine Screenshot.

Gruß Jollinar

Kalli01, 2023-08-17:
Hi,
bei mir kann ich ein ähnliches oder vielleicht auch das selbe Verhalten beobachten.
TruckiStick und Sun600. Zähler wird über Tasmota ausgelesen.

Heute Morgen gegen 5:30 Uhr konnte ich es live auf der Weboberfläche des Sticks beobachten.
ZEPC steht auf Target 50 und Average auf 5
Die Zähler Werte wurden noch aktualisiert. Der ACSetpoint war Konstant aber der ZEPC Wert ist immer mehr gestiegen. 
So als wollte er nach regeln aber es hatte keine Auswirkung auf den ACsetpiont
Das Powerlimit steht auf 450W

Habe dann erst die AC Spannung ausgeschaltet und beim 2. Versuch auch die Gleichspannung aber er wollte nicht mehr anfangen zu regeln.

Ein Logfile habe. Im Anhang noch ein Screenshot vom 15. Hier war es vermutlich auch so.
Wird die Leistung zurück geregelt wenn die Batteriespannung sinkt?
VBatCutoff steht auf 43V.  Die Spannung des Sun600 ist ca 1,2V niedriger als die an den Klemmen gemessene Spannung

[log (1).txt](https://github.com/trucki-eu/Trucki2Shelly-Gateway/files/12370881/log.1.txt)
[image]

Kalli01, 2023-08-17:
Hi Trucki

ja das wird es sein :)
hatte gleich am Anfang im Sun die Cutoff Spannung auf 47,5V und die Reboot Spannung auf 51V eingestellt.

Laut Grafana war die Spannung (Gemessen vom Sun) heute morgen bei 47,5V. Also kein Wunder warum er auch nach einem Neustart nicht mehr angesprungen ist.

Die Einstellung im Sun hatte ich total vergessen und habe die letzten Tage nur die Einstellungen im Stick geändert :D 

So Frage als Laie. Kannst du die Parameter ( wie die Cutoff und Reboot Spannung) auch aus dem Sun auslesen um auf der Weboberfläche eine Hinweis auszugeben? (Einstellungen unterschiedlich)
Falls nicht wäre vielleicht ein kleiner Hilfetext sinnvoll.

trucki-eu, 2023-08-17:
Nein, dass geht leider nicht. Das gibt die Schnittstelle nicht her.

Gruß Trucki

closed 2023-08-17 by trucki-eu