trucki-eu/Trucki2Shelly-Gateway #20: Keine Verbindung zum Lumentree 600

Pony1966, 2023-08-10:
ESP 8266 geflasht mit V1.12, Mqtt ok, Tasmota Einbindung Zähler ok, Sun fail keine Daten
Lumentree 600, V 6.1 ok, Gerät ist ca 1 Woche in Betrieb, 
Init ESP am PC geprüft über RS232 USB Adapter nach hterm.
Im Log wird angezeigt ERROR Modbus
Prüfung Lumentree 600 nach PC
 TX -> 01 03 00 46 00 01 65 DF Lumentree antwortet ok
 TX -> 01 04 00 07 00 01 80 0B fail timeout error

Alle Input Register 0X04 mit Modbus Poll geprüft, fail timeout error
Holding Register 0x03 sind meist lesbar,

Write 01 10 00 40 00 01 02 00 00 A8 90 scheint ok, Lumentree Ausgangsleistung 4 Watt

trucki-eu, 2023-08-11:
Ist der Lumentree aus dem Ask4It Shop? Nur diese haben die spezielle Firmware, die erlaubt die Leistung über den Remote-Port vorzugeben. Für alle anderen empfehle ich meine RS485 Platine, die den SUN über die analoge Schnittstelle regelt.

Pony1966, 2023-08-12:
Danke dir, der Lumentree 600 ist von Amazon, der Kanal halt sein Link.
Läuft nun, an der NodeMcu ist ein TTL - RS232 Adapter zum Lumentree angeschlossen.
Sehr gutes Projekt, Respekt.
Super Cool ist ZEPC, bin begeistert, nur schade das die Angebote für den LT 600 und die Antenne sehr rar sind.

Kalli01, 2023-08-12:
Hallo Zusammen.  Wirklich tolle Projekt. Der Stick mit dem Wechselrichter ist sehr flexibel.

Mein Lumentree 600 ist ca 2 Wochen alt. Nach dem Aufstecken des Sticks konnte ich in über Wlan einrichten aber der Wechselrichter wurde nicht erkannt.
Mir ist dann aufgefallen das der Stick nur Lose auf der Schnittstelle des Wechselrichters gesessen hat. Die Schrauben waren aber angezogen.
Nachdem ich die Muttern am Wechselrichter durch kürzere ersetzt habe, sitzt der Stick fest und die Verbindung zwischen Stick und Wechselrichter ist da.

trucki-eu, 2023-08-14:
Hallo, 
freut mich, dass es funktioniert. Ja, die zulangen 6-Kant Bolzen sind ein bekanntes Problem. In Zukunft wird der Hersteller kürzere Bolzen verbauen, aber das dauert eine ganze Weile, bis diese Wechselrichter in den Umlauf kommen.
Bis dahin kann man entweder: 
- von den Bolzen ca. 1mm abfeilen
- oder Online kürzere bestellen: UNC#4-40, 6-Kant-Höhe:5mm, Außengewindelänge: SUN1/2000:7mm, SUN600:5mm

Viele Grüße,
Trucki

closed 2023-08-14 by trucki-eu