trucki-eu/Trucki2Shelly-Gateway #21: Passwortgeschützte Tasmota-Oberfläche kein Connect

wattlooper, 2023-08-12:
Moin,
da ich ein Tasmota SML Device in My Fritz eingebunden habe, muss ich die Weboberfläche mit einem Passwort schützen.
Im Stick habe ich mehrfach das korrekte Passwort u. User eingetragen.
Nun findet der Stick leider mein Device nicht mehr.
Beste Grüße

trucki-eu, 2023-08-14:
Hallo Wattlooper,

ich habe das gerade bei mir getestet:
- Im Tasmota unter Configuration->Other->Web Admin Password ein Passwort gesetzt
-  in T2SG unter Meter-> Username: admin, password eingetragen
- ca. 30s gewartet ... funktioniert

Alternativ: 
- habe ich im T2SG->Meter Passwort leergelassen
- und bei Meter URL http://admin:passwort@192.168.1.141/cm?cmnd=status%2010 eingetragen

geht auch... 
Gruß,
Trucki

wattlooper, 2023-08-18:
Moin, beide Vorgehensweisen getestet .... leider kein Erfolg.
Tasmota 0.1.9.0.2

closed 2023-08-19 by trucki-eu

wattlooper, 2023-08-26:
{"WARNING":"Benötige user=<Benutzername>&password=<Passwort>"}
ist immer noch die Antwort von Tasmota

wattlooper, 2023-09-04:
Geht so aber:-)
http://192.168.178.40/cm?user=admin&password=1234&cmnd=status%2010