trucki-eu/Trucki2Shelly-Gateway #22: Das max. Powerpoint wird unabhängig vom Meter angefahren.

wattlooper, 2023-08-13:


wattlooper, 2023-08-13:
Irgend wie lief da was schief.... nun der Kommentar.
Nun ich kann anstellen was ich will aber die Steuerung schießt weit über ihr Meter- Ziel hinaus.
Hat zur Folge: Es wird mehr eingespeist als verbraucht wird.
Firmwrare 1.11 -Truckistick

trucki-eu, 2023-08-14:
Hallo Wattlooper,
auf Grund der Vielzahl an funktionierenden Installationen ist ein Konfigurationsfehler ist wahrscheinlicher als ein Bug.
Leider habe ich zu wenige Informationen, um Dir helfen zu können.
Könntest Du bitte hier einige Screenshots von deiner Konfiguration des T2SG posten.
Typische Fehler sind:
- Falsche Parameter: Z.b. sollten Target=30 und Average=10 sein.
- Shelly 3EM falsch herum eingebaut. Verbrauch vom Netz = positive Leistung, Export ins Netz= negative Leistung
- SUN an falscher Stelle eingebaut: Der SUN muss nach dem Meter eingebaut werden. D.h. Versorger->Zähler->Shelly 3EM->SUN . Wenn der SUN Leistung erzeugt, dann sinkt die gemessene Leistung am Shelly.

Viele Grüße,
Trucki

wattlooper, 2023-08-15:
Vielen Dank für die Antwort.
Also....Meine Konfiguration:
Zweirichtungszähler wird ausgelesen und auf dem Tasmota lief/läuft ein Script, welches meinen Soyosoucre und HLG 600 steuert. Es gibt keinen Broker!
Das hat prima, aber nur bei 350W max., funktioniert.
Mir war klar, das ich die Leistungs - Ansteuerung des Soyosource auskommnentiern muss.
Lediglich das 230V on/off,  um Leistungs - Überschwinger zwischen Ladeteil und WR zu vermeiden, habe ich im Script gelassen.
Bedeutet: Der Lummentree bekommt erst 230V auf die Nase wenn benötigt- leider mit einer neuen Bootsequenz.
Den Average habe ich aus einen früheren Post( vorher nicht gesehen ) auf 10 erhöht (obwohl auch schon mal durchgespielt) verändert. 
Nun läuft das erst mal aber noch nicht stabil.
Mein to-do: Script abändern, damit der WR dauernd Netz hat und gleichzeitig sicherstellen das das HLG nicht parallel läuft.
Und dann probiere ich die Lösung mit der verschlüsselten Verbindung.
LG Wattlooper

Anbei mein aktuelles Script welches ich noch eindampfen werde:

``>D 25

pcurr=0
pcurr2=0
sw=50
ofs=0
ofs2=0
dimmlevel=0
dimmlevel2=0
dimmlevel3=0
url_1="[192.168.178.61]"
url_2="[192.168.178.44]"

>B

=>sensor53 r

>T
pcurr=0-SML#Power_curr

>M 1
+1,3,s,16,9600,SML
1,77070100010800ff@1000,EVU Bezug,KWh,Total_in,4
1,77070100020800ff@1000,Einspeisung,KWh,Total_out,4
1,77070100100700ff@1,Aktueller Verbrauch,W,Power_curr,0

>S
pcurr2=0-pcurr

;print "pcurr2" %pcurr2%

ofs=(pcurr-sw)/180

if ofs<0{
ofs=-ofs
}

if pcurr>sw{
dimmlevel=dimmlevel+ofs         
} 
else{
dimmlevel=dimmlevel-ofs   
}

if dimmlevel>100 {
dimmlevel=100
} 
dimmlevel3=dimmlevel3+ofs

if dimmlevel<100 {
dimmlevel3=dimmlevel3-ofs
}

if dimmlevel<100 {
dimmlevel3=0
}
if dimmlevel3>100{
dimmlevel3=100

} 
;->WebSend %url_1% Channel3 %0dimmlevel3%
;print "dimm3" %dimmlevel3%

if pcurr<=0 {
dimmlevel=0
}
print %dimmlevel%

;=>dimmer %dimmlevel%
->WebSend %url_2% dimmer %0dimmlevel%

ofs2=(pcurr2-40)/180

if ofs2<0{
ofs2=-ofs2
}
Print "ofs2" %ofs2%

if dimmlevel2<0 {
dimmlevel2=0
}

if pcurr2>10{
dimmlevel2=dimmlevel2+ofs2        
} 
else{
dimmlevel2=dimmlevel2-ofs2   
}

print "ofs2" %ofs2%

if pcurr2>45 {
->WebSend %url_2% Power on
}
print "Dim2" %dimmlevel2%
print "Pcurr" %pcurr%

if dimmlevel2>55 {
dimmlevel2=55
}

;->WebSend %url_1% Channel2 %0dimmlevel2%

if pcurr>0 {
->WebSend %url_2% Power off
dimmlevel2=0
}

#

wattlooper, 2023-08-18:
Habe es hinbekommen. Danke!
Komisch hatte ich die selben Einstellungen schon mal und da ging es nicht.
LG

closed 2023-08-19 by trucki-eu