trucki-eu/Trucki2Shelly-Gateway #23: ZEPC Regelung LT600 - Shelly 3 EM - ZEPC bleibt bei 600 fest stehen - keine Regelung mehr (zu #19)

Jollinar, 2023-08-21:
Hallo Trucki,

jetzt habe ich den Fall wieder und kann dir auch ein paar Screen shots beifügen. 

Battery Voltage = 52,X
Cutoff Voltage im SUN = 44 Volt
Vbat cutoff = 45

also daran sollte es nicht liegen.
Der Shelly sendet korrekte Werte nur der die Regelung scheint nicht zu funktionieren.
ZEPC bleibt auf 600 W und Meter schwankt auf Grund der Verbrauches.

Ich habe dann den Lumentree 600 vom Netz genommen und wieder ans Netz angeschlossen und nun klappt die Regelung wieder scheinbar hängt der sich irgendwo im ZEPC auf und kriegt eine Festwert von 600 W gemeldet und berechnet nicht mehr den aktuellen Wert.

[image]
[image]
[image]
[image]

Gruß Carsten

Jollinar, 2023-08-22:
Hallo Trucki,

der Aufbau ist korrekt beschrieben, Zähler, Shelly und SUN. 

Ein Reboot des TSG-Stick hilft leider nicht, denn nach dem booten setzt der SUN alles sofort wieder auf max. Bisherige Lösung ist SUN vom DC trennen über die Sicherung und dann SUN vom AC trennen. Nach dem Neustart geht alles wieder wie gewohnt.

Cut-off habe ich jetzt auf 40V runtergesetzt aber eben erst nach dem letzten auftreten wobei die Option ja aus gegraut ist weil ich im SUN ja keinen Festwert in W hinterlegt haben.

"Was passiert, wenn Du in diesem Zustand dem SUN manuell eine Leistungsvorgabe über das Drop down Menu unter SUN1 gibst?" <-- kann ich gerade nicht testen da er nicht in dem Zustand ist.

Aktuell läuft die Regelung wieder.

Jollinar, 2023-08-24:
Moin Trucki,

anbei ein kleines Upate. Heute morgen ist es leider wieder passiert.

Habe dann AC SetPoint manuell gesetzt.

1. AC SetPoint > aktuelle Produktion <-- keine Wirkung 
2. AC SetPoint < aktuelle Produktion <-- Runterregelung auf AC SetPoint
3. AC SetPoint = autommatik <-- kurzfristig versuchte er zu Regeln ging dann aber wieder auf AC SetPoint = 600 

Erste Vermutung war Battery Voltage zu gering da sehr nah am Vbat-Cut off.

Dann kam auch der Cut-Off und das System stellte die Produktion ein, nachdem dann wieder VBat Reboot erreicht wurde, startet die Regelung wieder mit AC SetPoint = 600W also ohne Regelung.

Jetzt gerade hat die Regelung wieder den normal Betrieb begonnen AC SetPoint = ZEPC. 

Wovon hängt ab wann und wie e regelt, gibt es hier vielleicht Abhänigkeiten der Parameter Vbat cutoff, Vbat Reboot, ZEPC ?
Vielleicht wäre es sinnvoll ein Log zu haben das man aktivieren kann um den Verlauf der Parameter zu tracken ?

[image]

[image]

[image]

trucki-eu, 2023-08-24:
Hallo Carsten,

Viele Dank für die detaillierte Beschreibung. Für mich klingt das stark nach Cutoff im SUN. Der schaltet nämlich nicht hart ab, sondern reduziert erstmal die maximal mögliche Leistung. Den sollte man sowieso ganz abschalten. Zur Sicherheit schaltet das BMS ab, im täglichen Gebrauch das T2SG.
Loggen kann man prima mit MQTT. Da werden jede Menge Werte rausgeschrieben.

Viele Grüße Trucki

trucki-eu, 2023-08-24:
Komisch. Funktionieren deine Lüfter? Thermal throtteling ist der 2. Grund aus dem der Sun die Leistung reduziert.
Gruß Trucki

Jollinar, 2023-08-25:
Beide Lüfter laufen einwandfrei. 

Habe sogar die Stecker Ports geprüft ob beider Lüfter korrekt angeschlossen sind. Alles ok.

Heute morgen hatte ich dann wieder eine Aussetzer bei 

Battery Voltage 48,2V 
AC Setpoint = 300W automatisch im Sun gesetzt (Nachzeitmodus)
Regelung = keine

AC Setpoint auf 0 gesetzt und dann wieer ZEPC enabled, alles geht wieder. 

Ab welchem VBatCutoff + X fängt der Sun den an langsam runter zu regeln ?

VbatCutoff 45V
Battery Voltage = 48,2V
Differenz = 3,2V ???

trucki-eu, 2023-08-25:
Das T2SG kann sowohl Cutoff als auch Nachtmodus. Schalte mal alle automatischen Einstellungen (Cutoff, Timer) im Display des Suns ab.

Jollinar, 2023-08-25:
Die sind alle komplett aus, die Regelung läuft nur über den T2SG. Dennoch bleibt er manchmal hängen und ich kann dir nicht sagen warum.

Jollinar, 2023-09-04:
Hi Trucki, ich habe die Config einmal angepasst. und Folgenden Werte eingestellt.

SUN Cut-Off = 44V
SUN Reboot = 48V
Solar Bat = 600W (aktiviert)

TSG Cut-Off = 46V
TSG Reboot = 50V

Zudem habe ich das BMS umgestellt auf
Sec Cut-Off Low= 40V (2,5V)
Sec Cut-Off HIgh = 58,4V (3,65V)

Sowie den Laderegler auf
Max Current / Floating = 54,4 (3,4V pro Zelle)

die letzten Tage hatte ich keine Ausfälle mehr, werde das aber weiter beobachten. Ich gehe ich ggf. davon aus das die Vielzahl der Parameter sich hier ggf. gegenseitig blockiert und ausgehebelt haben.

Gruß Carsten

trucki-eu, 2023-09-04:
Hallo Carsten,

Vielen Dank für die Rückmeldung. Ich verstehe nicht, warum Du überhaupt einen Cutoff im SUN einstellst?

Viele Grüße Trucki

Jollinar, 2023-09-04:
Hi Trucki,

hatte ich vorher auch nicht aber da hatte ich dann ja die Probleme ich teste das jetzt mal und dann werden wir sehen.

anderer Frage wäre kann ich im Target bei ZEPC auch negative werte einstellen ?

Gruß Carsten

Jollinar, 2023-09-28:
Hi Trucki, kann geschlossen werden, der Langzeittest war erfolgreich mit den Einstellungen läuft es ohne Probleme.

Danke für die Hilfe.

closed 2023-09-28 by Jollinar