trucki-eu/Trucki2Shelly-Gateway #24: Erweiterungsanfrage: Gesamtleistung (kWh) vom SUN über MQTT

MrTree84, 2023-08-21:
Hi Trucki,

ich habe gestern schon mit Christian zu diesem Thema geschrieben und habe gedacht ich erstelle mal hier noch einen Eintrag.

Ich fände es super, wenn man die gesamte erzeugte Leistung des SUN auch über MQTT erhalten könnte. Dieser Wert wird ja, zusammen mit dem Tagesertrag, im Energiemenü des SUN angezeigt. 
Bei mir dienen die Werte zu Monitoringzwecken im SmartHome z.B. Ermittlung erzeugte Leistung (Tag, Woche, Monat usw.), Berechnung Einsparung und Amortisierung, Wirkungsgradberechnung DC ->AC, verschiedene Mittelwerte.

Aktuell muss ich leider einen Workaround mit einem Shelly PlugS machen, um an diesen Wert zu kommen.

Vielleicht ist es ja mit geringem Aufwand möglich den vorhandenen Wert auch über MQTT bereit zu stellen.
Würde sicher nicht nur mir bei der Integration ins SmartHome helfen.

Mein System: SUN2000G2 mit TruckiStick, SmartMeter mit Leistungswerte über dessen API, SmartHome Software inkl. Regelung des SUN über MQTT: IP-Symcon
Funktion bisher Fehlerfrei (allerdings erst 3 Tage wirklich am Netz ;-) )

Danke dir für die tolle Arbeit!

Grüße
Hoger

trucki-eu, 2023-08-22:
Hallo Hoger (Holger?),
ja die Frage kommt öfters. Auf dem Remote-Port an dem der Stick angeschlossen ist liegen auf den Registern  60 (Day active Power) und 63 (Total active power). Ich habe die gerade mal ausgelesen, bekomme aber auf die schnelle keine sinnvollen Werte, die zur Anzeige im Display passen. Werde ich mir bei Gelegenheit vielleicht mal ansehen.
Viele Grüße,
Trucki

MrTree84, 2023-08-22:
Hallo Trucki,

danke für die schnelle Rückmeldung und schade, dass es nicht auf Anhieb funktioniert hat.

Ich habe bei mir mal die Abfrage der Register über curl gemacht, aber leider gar keine Antwort erhalten :-(
Habe bei mir folgende url verwendet: $url = "http://$ipAddress/?modbus_reg=63";

Idee (kann, muss aber nicht):
Vielleicht kannst du deine Antwort und den dazugehörigen Zählerstand hier posten, vielleicht hat ja einer noch eine Idee, wie es zusammenpassen könnte. Oder auch die passende curl Abfrage, evtl. habe ich bei mir einen Fehler gemacht.

Danke für die Unterstützung.

Grüße
Holger

trucki-eu, 2023-08-22:
Hallo Holger,

ja so einfach geht das nicht. Mit dem "CURL Befehl" kannst Du nur schreiben, nicht lesen. Aktuell habe ich zuviele Nutzer die es nicht schaffen ihr WLan einzurichten, mit der englischen Github Seite nicht zurecht kommen, es nicht schaffen sich per USB Log auslesen selbst zu helfen, usw... Da müssen neue Features warten.

Viele Grüße,
Trucki

MrTree84, 2023-08-22:
Danke für deine Antwort.

Das kann ich absolut nachvollziehen. Wie bereits geschrieben kann man sich ja mit einem Shelly oder Co die Werte auch so ermitteln.

Ich hoffe das sich die anstehenden Schwierigkeiten gut beheben lassen und du wieder mehr Zeit für dich hast.

Danke.

Grüße
Holger

Cino2424, 2024-06-23:
Gibt es schon was neues zu dem Thema?

erxbout, 2024-06-23:
Gibt ein neues Update seit kurzem. Siehe #49