trucki-eu/Trucki2Shelly-Gateway #25: Regelung Einspeisung

budaustrade, 2023-08-21:
Ab einem gewissen Stromverbrauch ist die Einspeisung laut Shelly EM3 zu niedrig. Wie kann das sein? Habe dazu Screenshots gemacht.
[image]
[image]
[image]
[image]
[image]

Danke im voraus für eine Antwort.

Mit besten Grüßen

Budaustrade

trucki-eu, 2023-08-22:
Hallo,
die ersten beiden Fotos des T2SG sehen für mich gut aus. Die anderen Diagramme auf dem Kopf verstehe ich nicht.
Viele Grüße,
Trucki

budaustrade, 2023-08-22:
Hallo Trucki,

die Diagramme auf dem Kopf sind 24 Stunden Verbrauchs-Diagramme von Shelly EM3. Beim ersten Diagramm habe ich 125,72 Wh auf Phase A verbraucht, auf Phase B 0 Wh verbraucht, auf Phase B 228,77 Wh eingespeist, auf Phase C 115,43 Wh verbraucht, insgesamt habe ich 241,15 Wh auf alle 3 Phasen verbraucht und 228,77 Wh insgesamt eingespeist. Das passt ja zu den Einstellungen, die ich im T2SG gemacht habe.

Beim zweiten Diagramm habe ich 493,78 Wh auf Phase A verbraucht, auf Phase B 0,02 Wh verbraucht, auf Phase B 285,43 Wh eingespeist, auf Phase C 114,98 Wh verbraucht, insgesamt habe ich 608,78 Wh auf alle 3 Phasen verbraucht und 285,43 Wh insgesamt eingespeist. Das passt nicht mit den Einstellungen, die ich im T2SG gemacht habe. Hätte hier T2SG eigentlich nicht ca. 595 Wh einspeisen müssen?

Beim dritten Diagramm habe ich 830,63 Wh auf Phase A verbraucht, auf Phase B 0 Wh verbraucht, auf Phase B 371,21 Wh eingespeist, auf Phase C 47,02 Wh verbraucht, insgesamt habe ich 877,65 Wh auf alle 3 Phasen verbraucht und nur 371,21 Wh insgesamt eingespeist. Das passt auch nicht mit den Einstellungen, die ich im T2SG gemacht habe. Hätte hier T2SG eigentlich nicht die 700 Wh einspeisen müssen, die ich als max power eingestellt habe?

Ich habe auch jeden Tag so gewisse Spitzen im Stromverbrauch, da ich das Warmwasser mit einem Warmwasserboiler mache. Hier werden je nachdem ca. 1,8 kWh verbraucht, und trotz alledem speise ich in dieser Zeit nur zwischen 430 und 630 Wh ein, obwohl ich eben max power auf 700 Wh eingestellt habe. Woher kommen zudem noch die Unterschiede von den Wh, die ich einspeise? Also es wurde noch nie die eingestellten 700 Wh eingespeist, egal wie hoch der Stromverbrauch war. Gibt`s dafür eine Erklärung?

Bedanke mich im voraus für Deine Bemühungen und Antwort.

Beste Grüße

Budaustrade

trucki-eu, 2023-08-23:
Hallo Budaustrade,

vielen Dank für die Erklärung der Diagramme. Grundsätzlich: maxPower ist die maximale Momentanleistung des Inverters in [W]. Also keine [Wh] . - Falls das hier irgendeine Rolle spielt. 

Deine T2SG Einstellungen sehen richtig aus.

Es gibt natürlich bestimmte Bedingungen bei denen das T2SG den SUN nicht einspeisen lässt. Z.B. Wenn die Batteriespannung 1x unter den Cutoff gefallen und nicht wieder über VReboot gestiegen ist.

Bei großen geschalteten Leistungen sorgt der Average Wert dafür, dass das T2SG dem Hausverbrauch verzögert folgt.

Ganz genau findest Du das nur raus, wenn Du den aktuellen Hausverbrauch sowie die aktuelle Einspeiseleistung loggst. Z.B. mit MQTT.

Viele Grüße,
Trucki