trucki-eu/Trucki2Shelly-Gateway #28: MQTT - Home Assistant nur Nur lesene Daten möglich?

marcellofiore, 2023-09-03:
Hallo Trucki, leider habe ich festgestellt, dass ich keine Automatisierungen verwenden kann, um z.B. einen Targhet-Wert über MQTT zu setzen.. 
[image]
[image]

Nur die Sensoren sind zum Ablesen verfügbar. Irgendeine Idee, warum?

[image]
[image]

erxbout, 2023-09-05:
Ich versuche mal zu helfen:

Denke dass im Homeassistant MQTT Autodiscovery kein schreiben mit drin ist..

Bei mir hab ichs gelöst in dem ich einen Helfer angelegt habe:
[image]

Mit dem dann eine kleine Automatisierung:
[image]

Zum Ändern einfach eine 'Entities' Karte mit dem erstellten Helfer auf dein Dashboard setzen und dort dann den Wert eintragen.

Der wird dann nach 10 sekunden geschrieben (oder wieviel du einstellst, hab ich gemacht falls ich mich doch noch umentscheide)

trucki-eu, 2023-09-05:
Hallo erxbout,

ja korrekt! Das HADiscovery habe ich auf Wunsch eines Users eingebaut. Leider hat das mit den zu Schreibenden Werten nie richtig funktioniert und der User hat sich nie wieder gemeldet.

Ich schicke Dir gerne den Code wenn Du damit rumspielen möchtest. Schreib mir einfach ne Mail.

Viele Grüße,
Trucki

marcellofiore, 2023-09-06:
> Ich versuche mal zu helfen:
>
> Denke dass im Homeassistant MQTT Autodiscovery kein schreiben mit drin ist..
>
> Bei mir hab ichs gelöst in dem ich einen Helfer angelegt habe: [image]
>
> Mit dem dann eine [...]

Hallo erxbout, danke für dein Hilfe. Ich versuch es gleich das nach zu bauen, was aber mir noch nicht klar ist, wo soll ich die neue Wert für das Target reinschreiben, bestimmt in payload, ich sehe das aber nicht in dein Code...?

erxbout, 2023-09-06:
In den Helfer input_number.gridpowertarget

Mit der entities Karte kann man das auf einem dashboard machen

``` yml
type: entities
entities:
  - input_number.gridpowertarget
```

Ansonsten in devices and services -> helpers
Wichtig ist dass der Helper als Number definiert ist, dann ist die Anzeige am Dashboard dem entsprechend

erxbout, 2023-09-06:
Ich habe den Helfer mit der entities Karte auf meinem dashboard

``` yml
type: entities
entities:
  - input_number.mqttreactorpowertarget
```
Dort ändere ich manuell den Wert, dieser wird dann im payload geschickt

Der payload verträgt templates, deswegen die geschwungenen Klammern, also wird hier der Wert von der Entität geschickt.. das kann natürlich auch ein sensor sein oder jede andere Entität. 
Hier nur ein Beispiel:
``` yml 
payload: "{{ states('sensor.iphone_battery_level') | int }}" 
```

Sollte es ein fixer Wert sein der geschrieben werden soll dann wäre der code wie folgt:

``` yml
service: mqtt.publish
  data:
  qos: "0"
  topic: T2SGFirst/TARGETOVR
  payload: "15"
```

Hope that helps ;)

marcellofiore, 2023-09-07:
Super, es scheint alles zum Funktioniere, vielen Dank.

closed 2023-09-07 by marcellofiore