trucki-eu/Trucki2Shelly-Gateway #30: Battery voltage Anpassen

Selyria, 2023-09-15:
Hallo Ermstal super Projekt Wollte mal fragen ob es möglich wäre die Battery voltage so zu machen das diese anpassbar ist bei meinen Beiden Sun 1000 Wechselrichtern ist die Battery voltage um 0.2V zu gering was ein zu frühes abschalten nach sich zieht. Wäre schön wenn man da in der Software eine Änderung vornehmen könnte.

derspe, 2023-09-18:
Das wäre in der Tat hilfreich. Bei mir sind es 1.4-1.5V Differenz zum BMS meines Pylontech-Speichers und 1.3V zur direkt an den Schraubklemmen des WR gemessenen Spannung.

Der WR scheint ordentlich daneben zu liegen, somit wäre eine Offset-Einstellung im TruckiStick eine Super-Lösung. 
[image]

BayernJoJo, 2023-09-24:
Das würde mir auch sehr gefallen. Bei mir sind es auch immer zwischen 0,6V und 0,9V Differenz zum Victron Smart Shunt.
Ansonsten großes Lob, es läuft einwandfrei.

Gruß

jakehh, 2024-06-04:
Bei meinem SUN1000G2 (2024 Version) liegt die Spannung des Wechselrichters ebenfalls deutlich unterhalb der gemessenen Spannung von Victron Smart Shunt (ca. 1,1 - 1,5 V) bzw. unterhalb der gemessenen Spannung an den Eingangsklemmen des Wechselrichters (ca. 0,9 - 1,3 V). 
[image]

Die Werte vom Smart Shunt zu nutzen zu können wäre schon sehr hilfreich.

Danke und viele Grüße! 
Jake