trucki-eu/Trucki2Shelly-Gateway #31: Home Assistant, Enery Plance

marcellofiore, 2023-10-04:
Hallo Trucky, kannst du bitte beim nächste Update die MQTT Daten anpassen?
Were schön wenn die sensor in die Energy Plance zu sehen sind. 
Hier sind die Befehle, ich habe einen Sensor gebaut, der mir die Epever-Daten an Home Assistant sendet.
https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics
https://community.home-assistant.io/t/how-to-enable-power-consumption-statistics/327102/12

Die Funktion, mit der ich die Daten sende und die fehlenden jSON Key hinzufüge, löst das Problem :)

```
void sendMQTTEnergyTodayDiscoveryMsg() {
  // This is the discovery topic for this specific sensor
  String discoveryTopic = "homeassistant/sensor/"+uniqueID+"/energy_today/config";

  DynamicJsonDocument doc(2024);
  JsonObject dev;
  JsonArray identifiers;
  char buffer[1024];

  doc["name"] = uniqueID+"Energy Today";
  doc["stat_t"] = stateTopic;
  
  doc["stat_cla"] = "total";
  doc["native_value"] = energy_today;
  doc["last_reset"] = lastMsg;
  
  doc["uniq_id"] = uniqueID+"EToday";
  doc["unit_of_meas"] = "kWh";
  doc["dev_cla"] = "energy";
  doc["frc_upd"] = true;
  doc["val_tpl"] = "{{ value_json.energy_today|default(0) }}";

  dev = doc.createNestedObject("dev");
  dev["name"] = uniqueID;
  dev["model"] = "epeverMF";
  dev["sw_version"] = "1.0";
  dev["manufacturer"] = "MFDev";
  identifiers = dev.createNestedArray("identifiers");
  identifiers.add("MF"+uniqueID);

  size_t n = serializeJson(doc, buffer);
  client.publish(discoveryTopic.c_str(), buffer, n);
}
```

Wenn etwas brauchst, sag Besheid, kann ich gerne helfen.
Grüße, Marcello :)

Hattie164, 2023-10-06:
Pop

sabu2473, 2024-10-19:
dayenergy und totalenergy werden ab V1.13 über MQTT gesendet
https://trucki.de/t2sg