trucki-eu/Trucki2Shelly-Gateway #33: mqtt password: tooooo short

DeTaurus, 2023-10-25:
Meinst du das ernst? MQTT Passwort größe von 19 Zeichen? 
Nach dem Speichern des MQTT Passworts wird unter einstellungen nur noch 19 Zeichen angezeigt und es wird sich nicht mit dem Server verbunden.

Do you seriously? MQTT password size of 19 characters?
After saving the MQTT password, only 19 characters are displayed under settings and it will not connect to the server.

erxbout, 2023-10-26:
Funktioniert bei mir mit 13 Zeichen..?
Kann ich nicht reproduzieren..

Eventuell im log vom MQTT broker nachschauen was das Problem ist?
Oder und das log vom stick anschauen..

DeTaurus, 2023-10-26:
Ja schön, mein MQTT Passwort hat 60 Zeichen.

erxbout, 2023-10-26:
Achso das ist mir aus der Eingangsfrage nicht klar gewesen

Dann müsste das eventuell erweitert werden, ich schätze das wird aufgrund von Speicherplatz erstmal so gewählt worden sein und bis dato ist das niemandem negativ aufgefallen..

Unabhängig davon: Darf man fragen warum 60 Zeichen?
Wenn die MQTT Verbindung nicht verschlüsselt ist dann nutzen viele Zeichen doch auch nichts oder?
Ist die MQTT Verbindung verschlüsselt?