trucki-eu/Trucki2Shelly-Gateway #35: Trucki2Schelly mit ShellEM arbeitet nicht korrekt am Offgrid-Inverter

CrazyPit57, 2023-11-14:
Projektaufbau 

Part 1 (ok)
LiFePo4-Akku (6kw) mit Lumentree & Shelly **EM3** - Arbeitet, Regelung ok.

Part 2(Fehler)
LiFePo4-Akku (6KW) mit 2,5KW OffGrid-Inverter & Shelly **EM** (nicht EM3) zum Messen der Leistung aus dem OffGrid-Inverter & Lumentree um die Abgabeleistung zu erhöhen
Trucky (Lumentree) findet den ShellyEM zeigt die Leistung, welche vom Offgrid-Inverter geliefert wird an. Lumentree regelt aber nicht, liefert die eingestellte maximale Leistung konstant. Im Testaufbau läuft der Lumentree allerdings Netzzeitig am Hausnetz (nicht am Offgrid-Netz), damit er nicht in den Offgrid-Inverter zurückspeist. Soll erst wenn die Regelung funktioniert ans Off-Grid-Netz mit versorgen.
Sobald ich im Lumentree von Shelly EM zurück auf den Shelly EM3 umstelle, funktioniert die Regelung wieder. Json Keys werden sauber gesetzt. 

Anhängende Fotos zeigen 
Einstellungen Lumentree max 100Watt, ShellyEM (192.168.2.198) & Json Key
Meter ca. 50Watt, Lumentree liefert 105 Watt
Softwareversion (letzte)

Woran könnte es liegen?

CrazyPit57, 2023-11-14:
gelöst ???
nach erneutem (identischem) Projektaufbau funktioniert die Regelung. Merkwürdig. 
Polarität (Stecker gedreht) im OffGrid-Netz kann ich mir nicht vorstellen. dann würde der ShellyEM ja negative Werte liefern.

Wenn jemand eine Idee hat...

shake2k, 2023-11-14:
das ist der wert für die lüfter je nach temperatur zum beispiel damit bei meiner steckdose 400Watt ankommt muss ich 418 eingeben und er macht auch bissl mehr bei dieser anzeige dann

CrazyPit57, 2023-11-14:
> das ist der wert für die lüfter je nach temperatur zum beispiel damit bei meiner steckdose 400Watt ankommt muss ich 418 eingeben und er macht auch bissl mehr bei dieser anzeige dann

Danke für die Einschätzung. Das ist aber nicht der Grund. Wenn ich z.B. max Power auf 300 Watt einstelle, produziert der Lumentree auch diese, obwohl der EM nur ca. 50 Watt an Netzbezug misst. Bei Änderung der "max. Power", wird immer diese "max Power" eingespeist, unabhängig von der gemessenen Bezugsleistung. 

Bei Umstellung auf den EM3 setzt die Regelung ein. 
Zurück auf den EM führt wieder zur "max Power"....

Nachdem heute Nacht einmal ein korrektes Verhalten produziert werden konnte, wird heute über Tag bei mehreren versuchen immer wieder "max Power" erzeugt, wenn der Meter auf den EM eingestellt ist.

Übrigens: ein invertieren der Messung (negativer Leistungsbezug) führt jeweils sofort zum korrekten "0-Wert" für die Leistungsregelung. Wird das invertieren aufgehoben, stellt sich das ursprüngliche Verhalten ein.

Ich bin derzeit ratlos....

shake2k, 2023-11-16:
Also ich hab den Shelly 3EM da steht bei mir bei json keys total_power drinnen und unter Interval 500ms statt wie bei dir 250ms
haste den richtigen Lumentree mit logo vom Kanal usw? Zeigt der Shelly in der App negative Werte? Bin ja nicht der Profi hier aber ich versuch mal zu helfen :P

CrazyPit57, 2023-11-17:
> Also ich hab den Shelly 3EM da steht bei mir bei json keys total_power drinnen und unter Interval 500ms statt wie bei dir 250ms haste den richtigen Lumentree mit logo vom Kanal usw? Zeigt der Shelly in der App negative Werte? Bin ja nicht der Profi hier aber ich versuch mal zu helfen :P

Danke für die Rückmeldung.
Ja, beim Shelly 3EM ist der Json Key total_power. Das ist korrekt und bei mir auch so.
Problem tritt ja nur auf, wenn ich auf den Shelly EM aus der OffGrid-Versorgung wechsel. Schelly EM ist der Typ mit zwei Messkanälen. Der hat nach Literatur und auch in den automatich erzeugtem Json Key einen anderen Eintrag. Und der Lumentree ist original vom Kanal. Arbeitet ja auch am 3EM korrekt.

Angezeigte Power-Werte sind positiv. 
Wenn ich negative Werte provoziere, geht die Lumentree-Leiustung auch korrekter Weise auf 0 runter.