trucki-eu/Trucki2Shelly-Gateway #36: Home Assistant MQTT Verbindungsabriss nach Neustart

ChristianLaessig, 2023-11-20:
Hallo Trucki, 
mir ist aufgefallen, dass bei einem Neustart von HomeAssistant die Verbindung zum Trucki-Stick abreisst. Un die Verbindung wieder herzustellen, muss man in den Settings  Home Assistant Discovery de- und wieder reaktivieren. Auf der Recherche nach einer Lösung bin ich über folgende Anmerkungen gestoßen: 

_It’s the normal behavior of MQTT. When you publish a payload to a topic, all the clients connected to the broker and subscribed to that topic will receive the payload. The payload is not normally stored on the broker so any client that connects afterwards, and subscribes to the topic, receives nothing.

Home Assistant is an MQTT client and that’s why, after a restart, it receives nothing from topics that begin with “homeassistant” because their payloads (the device configurations you published previously) are no longer on the broker.

In contrast, if you publish the payload as a retained message, it instructs the broker to store it. Any client that connects and subscribes afterwards will receive the stored payload (i.e “retained message”).

**All devices that claim to support Home Assistant’s MQTT Discovery, should publish their discovery payloads as retained messages. If they don’t then you get the behavior you have observed.**_

Kannst du das bestätigen, vor allem den letzten Absatz? Wird es hierzu eine Lösung geben?

Danke und viele Grüße
Christian

pveredas, 2023-11-29:
Same issue here

jorismocka, 2024-04-14:
Hi Trucki,

I have the same issue here. Easiest way of a workaround, would be to have the option to reboot it via CURL.
Then I would add an automation if HA was restarted, restart the T2SG via curl too.

Is there an option to reboot T2SG via curl?

Thanks a lot, especially for your great work.

Regards
Joris

zwockel2023, 2024-05-18:
Hi Trucki,
I have the same problem. 
But only with Trucki2Shelly Gateway. 
This does not exist with Trucki2Meanwell Gateway.

Best Regards
Ralf

Sanke1, 2024-05-21:
Same issue, TS2S Stick disconnects and no more values to HA delivered, need to reboot the stick.

thwe1966, 2024-06-06:
Hallo Trucki,

auch ich habe das selbe Problem mit meinem SUN1000 G2 und deinem Stick Version: 1.12!

Wie bereits oben beschrieben:  "All devices that claim to support Home Assistant’s MQTT Discovery, should publish their discovery payloads as retained messages."

Ist das von deiner Seite aus nicht möglich? Dann wäre hier ein Hinweis doch angebracht, Danke!

trucki-eu, 2024-06-06:
> Hallo Trucki,
>
> auch ich habe das selbe Problem mit meinem SUN1000 G2 und deinem Stick Version: 1.12!
>
> Wie bereits oben beschrieben: "All devices that claim to support Home Assistant’s MQTT [...]

Hallo thwe1966(?),
na dann hier der "angebrachte" Hinweis:
- Ja ist möglich -
(und kommt mit der V1.13 in Kürze)
Gruß,
Trucki

thwe1966, 2024-06-06:
> Hallo thwe1966(?), na dann hier der "angebrachte" Hinweis:
>
> * Ja ist möglich -
>   (und kommt mit der V1.13 in Kürze)
>   Gruß,
>   Trucki

Danke für die Rückmeldung.

Warum Du Dir die Mühe machst nach meinem Nick ein Fragezeichen in Klammen zu setzen und auch noch das eine Wort in Hochkommas kann ich nicht nachvollziehen.
Sollte ich Dich mit meiner Wortwahl geärgert haben, dann tut es mir NICHT leid - immerhin waren schon einige vor mir mit dem Problem OHNE Antwort.

sabu2473, 2024-10-19:
solved with V1.13 
get here: https://trucki.de/t2sg

ickedude, 2026-09-28:
Moin,

ich bin gerade auf Version 1.16 und habe gerade mal die MQTT Nachrichten geprüft und kann das Retained Flag leider nicht sehen. Muss man noch etwas einstellen?

trucki-eu, 2026-09-29:
Nur die homeassistant config Topics sind Retained:

[image]

Gruß,
Trucki

ickedude, 2026-09-29:
Genau das sehe ich bei mir nicht:

[image]

Ich habe es aber heute morgen kurz gesehen. Ich prüfe mal meinen Broker.

ickedude, 2026-09-29:
Jetzt ist sie retained:

[image]

Wenn ein Gateway booted, dann schickt er die Nachricht raus und mqtt brower zeigt sie an. Wenn ich den mqtt browser neu starte, dann bekommt er vom broker die Nachricht geschickt, welche retained ist.
Es sieht mir nach einer Fehlinterpretation meinerseits aus. Ich bitte die Störung zu entschuldigen und bedanke mich für die schnelle Rückmeldung.

closed 2026-09-30 by trucki-eu