trucki-eu/Trucki2Shelly-Gateway #38: Request: Fronius API integration

DraycD, 2023-12-26:
Hi, i ch liebe deinen Stick!

wie kann ich meinen Fronius Smart meter über den wechselrichter einbinden?

API die gehen sollte wäre:

**http://fronius/solar_api/v1/GetMeterRealtimeData.cgi?Scope=System** und der Value sollte "PowerApparent_S_Sum" sein, aber wenn ich das in das "Meterfeld" eingebe bekomme ich nichts gelesen ;(

[image]

hat jemand ne idee was ich falsch mache?

Das kommt übrigens beim auslesen im browser - also bei http://192.168.2.7/solar_api/v1/GetMeterRealtimeData.cgi?Scope=System :
[image]

erxbout, 2023-12-26:
Hallo

Meinem Verständnis nach müsstest du im Json keys Feld das hier verwenden:
```
Body,Data,0,PowerReal_P_Sum
```

Der Stick sollte dann durch das Json richtig durchgehen (also Body dann Data ...)
Kann ich allerdings bei mir nicht nachstellen weil ich das Gerät dazu nicht habe, also keine Garantie auf Funktion..

Und warum PowerReal? 
Weil die anderen Werte sich auf Scheinleistung beziehen die man nicht wirklich "verbraucht" in dem Sinne..
Einfacher gesagt: Für ZEPC interessiert uns nur Wirkleistung (Also PowerReal) weil für die zahlen wir und die wollen wir kompensieren

Die Bezeichnungen sind meistens S, Q und P wobei P Wirkleistung ist

DraycD, 2023-12-26:
oh stimmt richtiger Wert ist "PowerReal_P_Sum"

erxbout, 2023-12-31:
Hats funktioniert?

DraycD, 2023-12-31:
leider nicht, habe es so eingetragen:
oben bei URL: **http://192.168.2.7/solar_api/v1/GetMeterRealtimeData.cgi?Scope=System**
unten bei JSON: **Body,Data,0,PowerReal_P_Sum**
kommt aber kein wert durch:
[image]

erxbout, 2024-01-13:
Also ich hab das mal versucht nachzustellen und es gibt anscheinend irgendwie ein Problem das JSON zu parsen..

mit dem minimalen JSON extract hier:
```
{"Body":{"Data":{"0":{"PowerReal_P_Sum":8}}}}
```
und dem oben genannten key bekomme ich keinen Wert ins Gateway:
[image]

Wenn ich aber das gleiche mache mit:
```
{"Body":{"Data":{"a":{"PowerReal_P_Sum":8}}}}
```
und den key dementsprechend auch ändere (also statt 0 ein a) dann bekomme ich einen Wert..
[image]

Kann hier leider das Problem nicht nachvollziehen.. maybe @trucki-eu hast du eine Idee?

trucki-eu, 2024-01-13:
Ich würde mal vorsichtig behaupten die " um die 0 gehören da nicht hin.

erxbout, 2024-01-13:
Das musst du leider Fronius sagen.. (Siehe screenshot in der allerersten Nachricht)

dg1yds, 2024-01-25:
Hallo,

ich habe zwar bisher nur den T2SG im Einsatz (nicht T2MW), diesen aber erfolgreich mit einem Fronius Wechselrichter/Smartmeter verbunden.

Dafür nutze ich aus der Fronius API einen andere Anforderung - ich nutze dafür die "GetPowerFlowRealtimeData" Schnittstelle; URL: "http://IP/solar_api/v1/GetPowerFlowRealtimeData.fcgi"
Damit sieht der notwendige String etwas anders aus: "Body,Data,Site,P_Grid".

Vielleicht probiert ihr es mal damit.

DraycD, 2024-01-25:
bei mir sieht die ausgabe so aus:

`{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 105,
               "E_Day" : 2835.400146484375,
               "E_Total" : 54334300,
               "E_Year" : 57908,
               "P" : 2320
            }
         },
         "Site" : {
            "E_Day" : 2835.400146484375,
            "E_Total" : 54334300,
            "E_Year" : 57908,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : -477,
            "P_Load" : -1843,
            "P_PV" : 2320,
            "rel_Autonomy" : 100,
            "rel_SelfConsumption" : 79.439655172413794
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-01-25T11:13:25+01:00"
   }
}`

DraycD, 2024-01-25:
Resolved - Danke euch

erxbout, 2024-01-25:
> Hallo,
> 
> ich habe zwar bisher nur den T2SG im Einsatz (nicht T2MW), diesen aber erfolgreich mit einem Fronius Wechselrichter/Smartmeter verbunden.
> 
> Dafür nutze ich aus der Fronius API eine andere Anforderung - ich nutze dafür die "GetPowerFlowRealtimeData" Schnittstelle; URL: "http://IP/solar_api/v1/GetPowerFlowRealtimeData.fcgi" Damit sieht der notwendige String etwas anders aus: "Body,Data,Site,P_Grid".
> 
> Vielleicht probiert ihr es mal damit.

Ah gute Idee!
Wusste nicht, dass der gewünschte Wert auch in einem anderen API Call existiert xD