trucki-eu/Trucki2Shelly-Gateway #39: Can not read shelly EM meter

pveredas, 2024-01-13:
log.txt created
Reset reason: Software/System restart
Trucki2Shelly Gateway V: 1.12 16.07.2023 06:56
{
  "admin_pass": "******",
  "mqtt_server": "192.168.*.*",
  "mqtt_port": "1883",
  "mqtt_user": "********",
  "mqtt_pass": "********",
  "hadiscovery": "1",
  "device_name": "T2SGF9FC73",
  "meter_url": "http://192.168.1.71/status",
  "json_keys": "emeters,0,power",
  "meter_interval": "500",
  "meter_invert": "0",
  "meter_user": "admin",
  "meter_pass": "",
  "maxPower": "1400",
  "vbatCutoff": "500",
  "vbatReboot": "505",
  "zepc_target": "60",
  "zepc_average": "10",
  "zepc_feedback": "0",
  "sun2_ip": "",
  "sun3_ip": "",
  "night_maxPower": "1",
  "night_begin": "16:00",
  "night_end": "07:00",
  "ntpserver": "pool.ntp.org",
  "timezone": "WET0WEST,M3.5.0/1,M10.5.0",
  "darktheme": "0",
  "ip": "192.168.1.119",
  "gateway": "192.168.1.1",
  "subnet": "255.255.255.0",
  "dns": "192.168.1.1"
}connected...yeey :)
Wifi IP: 192.168.1.119
Wifi GW: 192.168.1.1
Wifi SN: 255.255.255.0
Wifi DNS: 192.168.1.1
RSSI: 56 %
mqtt connected
mqtt subscribed to T2SGF9FC73/ACSETPOINTOVR
mqtt subscribed to T2SGF9FC73/DACOVR
mqtt subscribed to T2SGF9FC73/CALSTEPOVR
mqtt subscribed to T2SGF9FC73/METEROVR
mqtt subscribed to T2SGF9FC73/MAXPOWEROVR
mqtt subscribed to T2SGF9FC73/TARGETOVR
Http connected to meter: http://192.168.1.71/status

HTTP GET failed. http code: -1
Http get grid meter read failed: -2

Now listening at IP 192.168.1.119 UDP port 4210
Night starts at 16:00
Night ends at 07:00
NTP server IP: 193.136.152.72
NTP server time: (day) 10:36
Modbus: Connected to Trucki pcb
Modbus VGrid: 2340
Starting Modbus to SUN GTIL2 in mode: TRUCKI

pveredas, 2024-01-13:
http://192.168.1.71/status

{"wifi_sta":{"connected":true,"ssid":"*******","ip":"192.168.1.71","rssi":-66},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":true},"time":"10:41","unixtime":1705142510,"serial":1765,"has_update":false,"mac":"8CAAB505BFF9","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":2156.20,"reactive":-850.58,"pf":-0.93,"voltage":234.31,"is_valid":true,"total":21684839.1,"total_returned":170544.0},{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":234.31,"is_valid":true,"total":0.0,"total_returned":0.0}],"update":{"status":"idle","has_update":false,"new_version":"20230913-114150/v1.14.0-gcb84623","old_version":"20230913-114150/v1.14.0-gcb84623","beta_version":"20231107-164916/v1.14.1-rc1-g0617c15"},"ram_total":51064,"ram_free":35652,"fs_size":233681,"fs_free":154616,"uptime":6413}

erxbout, 2024-01-13:
Hello

I tried the json you provided, that worked for me.. (so the parsing seems fine the problem is the http connection..
Maybe its a problem with the authentication? 
Is the meter_user and the meter_password set correctly? (does the shelly even have a set password?)
Is the meter ip static?

Are you familiar with postman, insomnia or curl.. if so can you provide the timeline output that reproducing the http request gives you?

pveredas, 2024-01-13:
No authentication on the shelly EM.

The curl works without issues:

curl http://192.168.1.71/status
{"wifi_sta":{"connected":true,"ssid":"*****","ip":"192.168.1.71","rssi":-67},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":true},"time":"17:33","unixtime":1705167236,"serial":7694,"has_update":false,"mac":"8CAAB505BFF9","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":166.03,"reactive":-432.43,"pf":-0.36,"voltage":238.56,"is_valid":true,"total":21688167.8,"total_returned":170669.8},{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":238.56,"is_valid":true,"total":0.0,"total_returned":0.0}],"update":{"status":"idle","has_update":false,"new_version":"20230913-114150/v1.14.0-gcb84623","old_version":"20230913-114150/v1.14.0-gcb84623","beta_version":"20231107-164916/v1.14.1-rc1-g0617c15"},"ram_total":51064,"ram_free":36360,"fs_size":233681,"fs_free":154616,"uptime":31138}

pveredas, 2024-01-14:
Hi All,

Have done a factory reset with no luck.
Is there a way to retrieve the current log?

pveredas, 2024-01-17:
[image]
would appreciate any help troubleshooting this issue.

erxbout, 2024-01-17:
I am very sorry but I think I am not able to help you here.. 
From my tests using your provided json it works on my stick..
Only thing that comes to my mind is maybe a connection timeout?

How long does the shelly take to respond?

Have you considered to work around this issue by using the mqtt METEROVR?

pveredas, 2024-01-17:
that did the trick, thanks for the help :)

automation:
alias: Trucki_METEROVR
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.shelly_em_05bff9_power_0
condition: []
action:
  - service: number.set_value
    metadata: {}
    data:
      value: "{{(states('sensor.shelly_em_05bff9_power_0')|round())}}"
    target:
      entity_id: number.t2sgf9fc73_meter_override
mode: single