trucki-eu/Trucki2Shelly-Gateway #41: AC_Setpoint always ZERO with Lumentree-Sun800

TDatFD, 2024-02-11:
~~~~  Fehler / Issue~~~~~~~~~~~~~~~~~~~~~~~~
Der mit ZEPC korrekt errechnete Einspeiseleistung erscheint nicht (in Read-Back) von AC_Setpoint.
Der Sun800 speist weiterhin konstant mit seinen limitierten Einstellungen ein.
Auch eine manuelle Vorgabe von AC-Setpoint am T2SG wird nicht übernommen
AC-Setpoint bleibt immer "0" !

Auffällig: es wird auch keine Device-Temperatur (36°C) vom Sun800 angezeigt.

Im Web alle Anleitungen / Videos mehrfach gesichtet und verglichen
Der einzigste - nicht lösbare - Unterschied ist die nicht übertragene Temperatur vom Sun800  !

~~~~  betroffen ~~~~~~~~~~~~~~~~~~~~~~~~~~
Getestet am Sun800 mit T2SG Firmwaren 1.09 / 1.10 / 1.11 / 1.12   -> alle "betroffen"
Sun800: Konfiguration im Display Setup geprüft.. nichts offensichtlich falsches drin.

~~~ Vermutung ~~~~~~~~~~~~~~~~~~~~~~~~~~
* Da die Sun800 Temparatur "0" (ungültig..) ist wird AC_Setpoint" auf NULL geklemmt
   ..deutet auf fehlenden Datenpunkt hin, oder Kommunikation ist nicht vollständig korrekt
* der AC_Setpoint wird nicht zyklisch übertragen.
   ..ebenso: Kommunikation ist nicht vollständig korrekt

Gegenanzeige: "Grid-Voltage" ud "Battery Voltage" werden vom Sun800 zyklisch korrekt übertragen
..also Kommunikation grundsätzlich vorhanden.

~~~ Setup ~~~~~~~~~~~~~~~~~~~~~~~~~~
Shelly 3EM: Hauptzählerkasten, Leistungswerte plausibel. Works fine !
Lumentree Sun800:  "Limiter BAT or SOLAR limited power Mode = 50W"
                   "BAT Cut-Off 24V"     "BAT ReBoot 24,5V"
LiFePo4: SOC=>80%,  Spannung 26,2V

~~~ Betriebsfall ~~~~~~~~~~~~~~~~~~~~~~~~~~
* Probeaufbau nur mit Batterie an Phase 1
* Sun800 speist permanent mit ~50W auf auf Phase 1 ein.
* Im T2SG.Meter korrekt umgestellt auf "emeters,0,power"
* Übernommener Leistungswert Phase 1 stimmt mit Webansicht Shelly-3EM  überein 

Verbraucher an Phase1 : 149W
-50W (Sun800) = 99W  vom Shelly3EM
-10W ZEPC = 89W die nachgeregelt werden müssten..
-> der AC_Setpoint bleibt aber auf NULL

~~~ weiterhin probiert ~~~~~~~~~~~~~~~~~~~~~~
Sun800 auf "Limiter BAT or SOLAR limited power Mode = 600W"
    -> es wird konstant mit 600W ein gespeist, calculated ZEPC fällt auf 0W ab
         S800 wird nicht gedrosselt, es werden 450W zum Versorger rückgespeist

Sun800 auf "Limiter BAT or SOLAR limited power Mode = 800W"
    -> es wird konstant mit ~800W ein gespeist, calculated ZEPC fällt auf 0W ab
         S800 wird nicht gedrosselt, es werden ~650W zum Versorger rückgespeist

Die Umstellung auf "total_power" am Shelly3EM auch keine Reduktion am Sun800.

######################################
Falsche Einstellung von mir ?
Sun800 nicht vollständig unterstützt / Fehlerhaft ?

Danke / Thanks
Helmar

![AC-Setpoint_alwaysZERO_240211](https://github.com/trucki-eu/Trucki2Shelly-Gateway/assets/159693176/d25c97b6-82d9-4838-95ff-7a201f87ae20)
[Log_240211.txt](https://github.com/trucki-eu/Trucki2Shelly-Gateway/files/14232064/Log_240211.txt)
![PXL_20240211_090335371](https://github.com/trucki-eu/Trucki2Shelly-Gateway/assets/159693176/33132135-2275-47f4-b2e6-9c80ea79488b)
![PXL_20240211_105616395](https://github.com/trucki-eu/Trucki2Shelly-Gateway/assets/159693176/af1c9d30-cde0-4ba4-8b53-6f0d11770702)
![PXL_20240211_110658916](https://github.com/trucki-eu/Trucki2Shelly-Gateway/assets/159693176/34e25d24-2748-43f7-9b19-727bf91d3227)

TDatFD, 2024-02-11:
Das Bild "PXL_20240211_090335371" - ziehe ich zurück: war lediglich ein Versuch, der auch nix  brachte..  hier falsch eingestellt..
Aktuell (wie original) wirksam: PXL_20240211_123105583.  Problem besteht weiterhin.
[image]

trucki-eu, 2024-02-11:
Hast Du einen Trucki2Sun Stick aus dem Ask4IT Shop? Nur die Lumentree SUN Inverter aus dem Ask4IT Shop haben die Spezialfirmware die erlaubt die Leistung über die DB9 Schnittstelle zu steuern... und die Temperatur auszulesen.

Entweder einen SUN von Ask4IT besorgen, oder die Trucki RS485 Platine mit Wifi Modul verwenden.

TDatFD, 2024-02-11:
T2SG Stick von Ask4IT:  habe ich, ist ja über AMA gut verfügbar.

Inverter: bisher keine "must haves" gefunden. Auch Ask4IT wirbt ja nicht mit dem Argument - deswegen nie drauf aufmerksam geworden..

trucki-eu, 2024-02-11:
Abgesehen, dass der Ama Shop nicht meiner ist, steht in der Artikelbeschreibung vom Stick: 

...funktioniert ausschließlich an "LumenTree featuring Trucki"-kompatiblen SUN-600G2-M/SUN-1000G2-M/SUN-2000G2-H

Wie geschrieben funktioniert die TruckiRS485Platine+Wifi Modul mit allen SUN G2.  In den 600er passt sie aber mechanisch nicht rein.

Gruß Trucki

TDatFD, 2024-02-17:
Gelöst mit "LUMENTREE Feat. Trucki Kontroll-Platine Upgrade-Kit für ältere SUN.." von AMAZON
Der vorhande Sun800 musste dazu **zusätzlich** in den Grundeinstellungen auf Sun1000 umgestellt werden.
WORKS JUST FINE !

Nebenfeststellung: der Sun800 ist lediglich ein per Firmware-Einstellungen gedrosselter 1000er - nicht nur als Vermutung..
Den durfte ich ja öffen: alle Elektronikplatinen inkl. Controller tragen inder Bezeichnung direkt die 1000 oder sind identisch mit dem 1000er. Selbst der über AMA bezogene modifitierte Controller hat die gleiche Revisionsnummer.

erxbout, 2024-02-17:
Spannend, cool dass das funktioniert ;)

Issue könnte man dann ja schließen?

Nebenfeststellung von mir: Beim Benutzen von einer 24V Batterie bekommt man AC seitig nur 850W maximal raus also sind wir ja nicht so weit weg von den 800 und man kann ja extra nochmal einstellen, dass 800W max ist ;)

Viel Spaß mit deinem Gerät :D

closed 2024-02-17 by TDatFD