trucki-eu/Trucki2Shelly-Gateway #43: MQTT values not available in Home Assistant after updating/restart

Spawny2015, 2024-03-03:
If HA needs restart, because of an update or like this. Values not available anymore. Needs to go to the Stick Menu to MQTT, deactivate settings for HA, then press save. After this activate the settings again for HA, press save and the Values are all available again.... Had this just for the Trucki Stick. All other MQTT values are available after restart.... I use an Mosquito MQTT....

opencanoe, 2024-03-21:
I have the same problem too. Nice workaround with deactivating the HA settings. I did a reboot everytime the problem occurs.

Spawny2015, 2024-03-25:
Better to insert this into configuration.yaml
Works now great! Just copy and insert your Stick-Name instead of ......
You will find then the entities under their name.

#Trucki2Shelly Gateway

    - state_topic: "T2SG....../ACSETPOINT"
      name: "T2SG_ACSetpoint"
      unit_of_measurement: "W"
    - state_topic: "T2SG....../ACDISPLAY"
      name: "T2SG_ACDisplay"
      unit_of_measurement: "W"
    - state_topic: "T2SG....../VGRID"
      name: "T2SG_VGrid"
      unit_of_measurement: "V"
    - state_topic: "T2SG....../VBAT"
      name: "T2SG_VBat"
      unit_of_measurement: "V"
    - state_topic: "T2SG....../DAC"
      name: "T2SG_DAC"
    - state_topic: "T2SG....../CALSTEP"
      name: "T2SG3_Calstep"
    - state_topic: "T2SG....../TEMPERATURE"
      name: "T2SG_Temperature"
      unit_of_measurement: "°C"
    - state_topic: "T2SG....../METER"
      name: "T2SG_Meter"
      unit_of_measurement: "W"

closed 2024-03-25 by Spawny2015