trucki-eu/Trucki2Shelly-Gateway #45: Steuerung des Trucki2Shelly-Gateway unter Home Assistent

Leon2079, 2024-04-21:
Hallo zusmmen,

ich bräuchte euere hilfe bezüglich der Stererung unserer Trucki2Shelly-Gateway Platine unter Home Assistent.

Ich wollte unter Home Assistent gerne die Trucki2Shelly-Gateway Platine mit den ACSetpointOVR befehlt steuern aber ich weiß nicht genau wo ich genau ansetzten soll.
Ich habe die paar code Zeilen aus der dukomentation rausgeschrieben aber steuern kann ich es über Home Assistent nicht.
Mit den IoBroker war das kein problem. 

Mfg
leon

erxbout, 2024-04-21:
Hallö

Ich habe das so gelöst, dass ich einen Helfer und eine Automation erstellt habe..

Der Helfer ist vom typ number und hat bei mir eine Range von 0-400 (natürlich beliebig wählbar aber große range als slider zb unpraktisch)

Die Automation schaut auf diesen Helfer (also bei Änderung aber erst wenn sich 10 Sekunden nix ändert weil ich vielleicht noch draufkomme einen anderen Wert haben zu wollen)
Sendet draufhin diesen payload:
``` yml
service: mqtt.publish
data:
  qos: "1"
  topic: BalkonyMKII/TARGETOVR
  payload: "{{ states('input_number.powertarget') | int }}"
```

Hope that helps ;)

Leon2079, 2024-04-22:
Hallo erxbout,

könntest du es vielleicht genau erklären was genau und wo genau im Home Assistent eingetragen werden soll.
Bin in dieser Sache nicht so bewandert und würde mich über genau nachahmungs Beschreibungen freuen.

MfG
Leon

erxbout, 2024-04-22:
#### Für den Helfer:

Settings -> Devices & Services -> (Tab ganz rechts) Helpers
Button rechts unten "+ Create Helper" -> Number auswählen -> Namen und Metadaten vergeben

#### Für die Automation:
Settings -> Automations & scenes -> Button rechts unten "+ Create automation" -> Create new automation
Ums für mich einfacher zu machen dann rechts oben auf die drei punkte und "Edit in yaml" auswählen und das hier reinwerfen:

``` yaml
alias: <name hier einfügen>
description: ""
trigger:
  - platform: state
    entity_id:
      - input_number.<name vom number helfer statt leerzeichen _>
    for:
      hours: 0
      minutes: 0
      seconds: 10
condition: []
  - service: mqtt.publish
    data:
      qos: "1"
      topic: <MqttBasisTopicVomGateway>/ACSETPOINTOVR
      payload: "{{ states('input_number.<name vom number helfer statt leerzeichen _>') | int }}"
mode: single
```

Den Helfer dann auf ein Dashboard werfen und gut ist.. eventuell die duration einstellen für den trigger wenn mans sofort haben will..
Es sollte gehen in der Automation wieder auf den GUI modus zurück zu wechseln, beim mqtt publish wird trotzdem noch yaml drinstehen..

#### Empfohlene Videos zum Thema:
- [simon42 - Home Assistant Template Sensoren und Hilfsentitäten (Helfer) 101](https://www.youtube.com/watch?v=tgZqukb2jHo&pp=ygUOc2ltb240MiBoZWxmZXI%3D)
- [simon42 - Home Assistant Automation Tutorial - Komplett & verständlich erklärt](https://www.youtube.com/watch?v=SGlkzKvJ96s&pp=ygUSc2ltb240MiBhdXRvbWF0aW9u)

Leon2079, 2024-04-23:
@erxbout 

Super es hat Funktioniert vielen vielen Dank.

Mein Fehler war das ich bei der erstellung der Automatisierung unter Entitaten die Nummerischen Zustand ausgewählr habe anstett nur den Zustand. Danach dein YAML code hinzufügen und anpassen und es klappt 👍 👍 🥇

Welwere, 2025-03-21:
> #### Für den Helfer:
> Einstellungen -> Geräte & Dienste -> (Tab ganz rechts) Helfer Button rechts unten „+ Helfer erstellen“ -> Nummer auswählen -> Namen und Metadaten vergeben
> 
> #### Für die Automatisierung:
> Einstellungen -> Automatisierungen & Szenen -> Button rechts unten „+ Automatisierung erstellen“ -> Neue Automatisierung erstellen Ums für mich einfacher zu machen, dann rechts oben auf die drei Punkte und „In Yaml bearbeiten“ auswählen und das hier reinwerfen:
> 
> alias: <name hier einfügen>
> description: ""
> trigger:
>   - platform: state
>     entity_id:
>       - input_number.<name vom number helfer statt leerzeichen _>
>     for:
>       hours: 0
>       minutes: 0
>       seconds: 10
> condition: []
>   - service: mqtt.publish
>     data:
>       qos: "1"
>       topic: <MqttBasisTopicVomGateway>/ACSETPOINTOVR
>       payload: "{{ states('input_number.<name vom number helfer statt leerzeichen _>') | int }}"
> mode: single
> Den Helfer dann auf ein Dashboard werfen und gut ist.. Eventuell die Dauer für den Trigger einstellen, wenn mans sofort haben will.. Es sollte in der Automatisierung wieder auf den GUI-Modus zurück zu wechseln gehen, beim mqtt-Publish wird trotzdem noch yaml drinstehen..
> 
> #### Empfohlene Videos zum Thema:
> * [simon42 - Home Assistant Template Sensoren und Hilfsentitäten (Helfer) 101](https://www.youtube.com/watch?v=tgZqukb2jHo&pp=ygUOc2ltb240MiBoZWxmZXI%3D)
> * [simon42 - Home Assistant Automation Tutorial - Komplett & verständlich erklärt](https://www.youtube.com/watch?v=SGlkzKvJ96s&pp=ygUSc2ltb240MiBhdXRvbWF0aW9u)

> #### Für den Helfer:
> Einstellungen -> Geräte & Dienste -> (Tab ganz rechts) Helfer Button rechts unten „+ Helfer erstellen“ -> Nummer auswählen -> Namen und Metadaten vergeben
> 
> #### Für die Automatisierung:
> Einstellungen -> Automatisierungen & Szenen -> Button rechts unten „+ Automatisierung erstellen“ -> Neue Automatisierung erstellen Ums für mich einfacher zu machen, dann rechts oben auf die drei Punkte und „In Yaml bearbeiten“ auswählen und das hier reinwerfen:
> 
> alias: <name hier einfügen>
> description: ""
> trigger:
>   - platform: state
>     entity_id:
>       - input_number.<name vom number helfer statt leerzeichen _>
>     for:
>       hours: 0
>       minutes: 0
>       seconds: 10
> condition: []
>   - service: mqtt.publish
>     data:
>       qos: "1"
>       topic: <MqttBasisTopicVomGateway>/ACSETPOINTOVR
>       payload: "{{ states('input_number.<name vom number helfer statt leerzeichen _>') | int }}"
> mode: single
> Den Helfer dann auf ein Dashboard werfen und gut ist.. Eventuell die Dauer für den Trigger einstellen, wenn mans sofort haben will.. Es sollte in der Automatisierung wieder auf den GUI-Modus zurück zu wechseln gehen, beim mqtt-Publish wird trotzdem noch yaml drinstehen..
> 
> #### Empfohlene Videos zum Thema:
> * [simon42 - Home Assistant Template Sensoren und Hilfsentitäten (Helfer) 101](https://www.youtube.com/watch?v=tgZqukb2jHo&pp=ygUOc2ltb240MiBoZWxmZXI%3D)
> * [simon42 - Home Assistant Automation Tutorial - Komplett & verständlich erklärt](https://www.youtube.com/watch?v=SGlkzKvJ96s&pp=ygUSc2ltb240MiBhdXRvbWF0aW9u)

Hallo 
Da ich auch noch keine Erfahrung habe Frage ich mal welcher Name muss da rein.
Mfg

Welwere, 2025-04-02:
Ja würde ich auch wissen alias:name