trucki-eu/Trucki2Shelly-Gateway #46: cannot get 100% output (600W)

madejackson, 2024-05-03:
When my T2SG is requesting 100% / 600W Output, it does only deliver about 560-570W

Same behaviour with manual MQTT or ZEPC.
I already did a calibration (set CAL to 1).

Is there any way to squeeze out the last 30-40W?

[image]

erxbout, 2024-05-03:
What is your Voltage supply?

My sun 1000 reports a power limit of 860W because of the supplied 24-27V
On 24V even lower (battery goes empty)

Also is your measuring equipment calibrated correctly?

madejackson, 2024-05-03:
The values all come from T2SG directly (via mqtt)
The solar input is higher than 600W at the same time, so that's why i wanna also output the max 600W. Rest goes into the battery, which is 26V right now.

[image]

trucki-eu, 2024-05-04:
Hast Du einen SUN600 oder 1000? Model 2023 oder 2024? Trucki RS485 Platine oder T2SUN Stick?

madejackson, 2024-05-04:
Es ist ein SUN600 2023 mit T2SG

trucki-eu, 2024-05-04:
Ja, dann ist das so. Ich wüsste nicht, was man da noch einstellen könnte. Gruß,
Trucki

closed 2024-05-05 by madejackson