trucki-eu/Trucki2Shelly-Gateway #47: T2SG mit Volkszähler

rt1970, 2024-05-06:
Da mein Shelly 30W Verbrauch zu wenig anzeigt (warum auch immer) möchte ich die Werte aus dem Volkszähler nutzen.
Leider regelt dann der Lumentree wild hoch und runter.
Der Grund scheint die zu langsame Abfrage am Volkszähler zu sein.
Kann das sein? Gibt es trotzdem eine Möglichkeit die beiden zu koppeln?

Grüße

rt1970, 2024-06-17:
Also... Hier meine Lösung:
erst dachte ich dass die Antwortzeiten zu hoch seien. Diese lagen aber bei ~250ms.
Dann habe ich mein Script angeschaut und geändert ("Aktuelle " gelöscht).
mein Script alt:
1,77070100100700ff@1,Aktuelle Leistung,W,Leistung,0
mein Script neu:
1,77070100100700ff@1,Leistung,W,Leistung,0
Damit läuft der Lumentree nun sauber und wird endlich direkt vom Zähler gesteuert.

PS: Trucki: Suuuuper Arbeit! *RESPEKT*

trucki-eu, 2024-06-17:
Vielen Dank für dein Lob! Freut micht, dass Du Freude damit hast!
Gruß,
Trucki

wnetx, 2024-06-17:
Ich habe die 1.13 geflasht (via WEB-IF) - Ich kam von der 1.12.
Da lief glaube ich etwas schief.
Nach 5min ohne Reaktion über Webinterface (aber immer noch pingbar) habe ich mal neugestartet (Strom Weg)
Es schickt brav die Daten per MQTT -> Komme ich noch anders an das Gerät -> Denke da hängt nur der Webserver ?
Reset / Neu flashen ?
Über einen Tipp würde ich mich freuen ;-)

trucki-eu, 2024-06-18:
Bei der Trucki Platine kannst Du das Log File über USB auslesen: 

https://github.com/trucki-eu/Trucki2Shelly-Gateway?tab=readme-ov-file#serial-debug-interface