trucki-eu/Trucki2Shelly-Gateway #50: Einrichtung Home-Assistant über T2SG & T2MG

PatFie2001, 2024-06-23:
Guten Tag,
ich bin neu in der Welt von Trucki und Home Assistant. 
Ich habe mit einen HA-Server besorgt, nun möchte ich die beiden Geräte T2SG und T2MG hinzufügen. 

Ich verstehe nicht so ganz, was ich bei den beiden Geräten an Daten eingeben muss. Augenscheinlich mache ich etwas falsch, da in HA keine neuen Geräte angezeigt werden. 

[image]

- die IP vom HA habe ich aus meiner Fritzbox.
-  als Port nehme ich den klassischen "8123" von HA an
- username von HA-Konto
- Passwort von HA-Konto

Was mache ich falsch? 

Brauche ich noch einen Separaten MQTT broker? 

Danke und viele Grüße,
Patrick

trucki-eu, 2024-06-24:
Hallo Patrick,
Du musst in deinem HomeAssistant unter Einstellungen->Geräte u. Dienste->MQTT installieren. Ggf. hierzu noch mal separat googlen. MQTT läuft dann auf dem Port 1883.

D.h. im T2SG stellst Du als Broker IP die HomeAssistant IP ein. Als Broker Port 1883. Den Haken Readonly setzen, wenn HomeAssistant nur lesen können soll und HA Discovery setzen, damit Du in "HomeAssistant-> Einstellungen->Geräte u. Dienste->MQTT->Geräte->T2SG->Zum Dashboard hinzufügen"  Das T2SG ohne weitere yaml Einträge finden kannst.

Viele Grüße,
Trucki

PatFie2001, 2024-06-24:
Hallo Trucki,
lieben Dank für die schnelle Hilfe! Hat alles funktioniert und sowohl der T2SG wie auch der T2MG funktionieren problemlos. 
die Einbindung in HA ist genial, die Visualisierung soll mir helfen das System besser zu verstehen und die Geräte  auf meine Bedürfnisse perfekt einzustellen. 

Viele Grüße 
Patrick

closed 2024-06-24 by PatFie2001