trucki-eu/Trucki2Shelly-Gateway #55: No MQTT connection possible

Phoeniix, 2024-10-06:
Hi,
I've set up my trucki stick the last days, so it is connected to my Wifi.
Unfortunately the MQTT connection isn't working until now. (broker is in the same network and reachable)
I have added all needed credentials and broker IP multiple times, but the trucki stick tells mit "mqtt failed: 5" in the log.
Also a reboot can't change anything.

Other MQTT clients can connect easily with the same credentials.

Is this a known issue? Anyone else facing this issue?
Is there a possibility to get more detailed information about the broken mqtt connection?

Thanks in advance!

Phoeniix, 2024-10-09:
After a few investigations it seems that my MQTT passwort (20 chars) was too big.
I've created also a user with a 40 char password, also no connection possible. (Broker Log: Client not authorized)
But if I shorten the password to 10 chars the connection can be established.
Maybe it helps someone else with the same problem.

closed 2024-10-09 by Phoeniix

erxbout, 2024-10-13:
same problem as #33

would be a small quality of life improvement to have it shown as hint in the web ui or something

trucki-eu, 2024-10-13:
see T2SG manual on https://trucki.de/t2sg page 10:
[image]