trucki-eu/Trucki2Shelly-Gateway #57: Batterie zwingend erforderlich?

harambe88, 2024-12-10:
Ich betreibe einen Victron 100/35 Solar Batterycharger in Verbindung mit LiFePo-Batterien (24V) und einem Sun1000G (2024)  mit Trucki2Shelly Gateway.

Die Nullsteuerung funktioniert ganz gut solange genug Saft vom Dach kommt. In den sonnenarmen  Monaten reicht die Leistung jedoch nicht aus, um die Leistung nur in den Inverter zu leiten. Es wird also zunächst immer die Batterie geladen und dann wird eingespeist. 

Ich habe das jetzt so eingestellt, dass bei Erreichen von 20 % der Batteriekapazität der Inverter zugeschaltet wird, darunter wird nur die Batterie geladen.  Die Steuerung läuft über Home Assistant. Also ständiges Laden/Entladen der Batterie. Dabei geht natürlich viel der ohnehin geringen Energie verloren.

Ich würde also am liebsten  während der Wintermonate die Batterie ganz abklemmen und alles direkt einspeisen.

Mit den Steuerungsparametern  bin ich trotz langem Probieren nicht weitergekommen.

Ich hoffe, dass dies der richtige Platz für mein Anliegen ist. Ein Trucki-Forum habe ich nicht gefunden.

erxbout, 2024-12-10:
Solange der Victron mit im Bunde ist hat man prinzipiell fast den gleichen Verlust..
PV DC -> 95% 24V -> 99% Speichern -> 95%AC = ca 89% Effizienz

Wenn ein Umbau denkbar ist, dann könnte man (sofern die Spannung der Panels zum Eingang passt also max 65V) die Panels direkt am WR anschließen und den einfach auf mppt umstellen..
PV DC -> 95% AC = 95% Effizienz..

Ist aber dann jedes Jahr ein Umbau, muss man sich überlegen wie sehr sich das auszahlt..
Und dann sind halt gute Tage wos Überschuss gibt potenziell, aber das ist Vebrauchsabhängig..

harambe88, 2024-12-10:
Gute Idee, meine Panels haben aber leider 72+ V. 

Mit welcher Einstellung kann man denn den Inverter auf MQTT umstellen?

erxbout, 2024-12-10:
MPPT.. Maximum Power Point Tracking..
Ich schätze, dass wenn man den AC SETPOINT auf 1000 stellt wird der WR so viel einspeisen wies geht..

Die Panels parallel schalten dann bleibt die Spannung bei 36V und der Strom erhöht sich..
Wenn ich die Daten vom Victron nehme, dann nehme ich an, dass um die 840W mit 2 panelen drauf sind?

Das wäre im Falle von Parallelschaltung ein max strom von 23A bei 36V

Müsste so gehen..

harambe88, 2024-12-10:
Vielen Dank. Das werde ich ausprobieren.
LG

trucki-eu, 2024-12-11:
Ja, die Anfrage den SUN mit T2SG mit Panels statt Batterie zu betreiben gab es schon mal. Soweit ich mich erinnern kann hat der ZEPC Probleme bekommen, wenn die Panels nicht ausreichend Leistung geliefert haben. Ich habe dafür damals den Haken: ZEPC->Use Display Feedback eingebaut. Allerdings hat sich der Tester dann eine Batterie besorgt und ich habe nie Rückmeldung bekommen.
Gruß,
Trucki

harambe88, 2024-12-11:
Ich habe jetzt probiert: 

- AC-Setpoint = 1000W  und 
- ZEPC->Use Display Feedback
- Batterie full und normal auf 23 V, geliefert werden vom Controller ca. 28V

Beides ändert nichts. Solange die Batterie offline ist, wird nichts eingespeist. Momentan ist es mit angeschlossener Batterie so, dass 33 W erzeugt werden, 5W gehen in den WR und 28 W in die Batterie. 

Wenn die Batterie abgehängt wird, kommen noch 5 W. Die gehen dann in den WR. In diesem Zustand zeigt der Charge Controller Absorption an.

Ich habe den Charge Controller auf LiFePo Batterie eingestellt. Vielleicht hilft da eine andere Einstellung. Wüsste aber nicht wo/wie.

erxbout, 2024-12-11:
Ich würde den Victron ganz rausnehmen und PV direkt an WR anschließen

Damit kann man dann leichter den WR einstellen, dass der mal richtig arbeitet + bessere Effizienz

Der Victron ist halt auch ein Batterieladegerät und wird komische Dinge machen in so einer Situation..

Bitte wenn möglich auch jemanden aus dem Umfeld auch dazuholen der sich auskennt, sicher ist sicher und 4 augen sehen mehr als 2..

erxbout, 2024-12-11:
Ich hoffe, dass diese Panels nicht grob über 1400W produzieren könnten weil sonst wär evtl der Victron sowieso zu klein..
Man muss hierbei bedenken, dass die 35A Ladestrom des Victron auf der Batterieseite sind.. im Zweifelsfall 24V*35A = 840W
Auf die ursprüngliche Frage ob die Batterie zwingend erforderlich ist: Vermutlich wegen dem Victron ja, dem SUN ist die Eingangsspannung egal solangs im Spannungsbereich vom SUN bleibt..

Also:

Der normale Aufbau ist:
_PV ~90V -> Victron MPPT ~90V auf 24V Batteriespannung -> Sun WR 24V Batterie auf 230V AC_

Der Victron ist darauf ausgelegt eine Batterie zu laden die eine eingestellte Spannungskurve zu folgen (für Batterie LiFePO4) für 24V Spannungen verdoppeln:
[image]

Falls der Victron eine Spannung auserhalbs dieses Bereiches erreicht, dann möchte er nicht mehr laden weil dann entweder ein Fehler der Batterie vorliegt oder die Batterie voll ist (Absorbtionsladen ist für das Abschließen des Ladevorgangs.. hier wird die Spannung gleich gelassen und der Strom wird immer weniger)

Sollte keine Batterie vorhanden sein, weiß der Victron das nicht und geht evtl auf Störung oder auf Absorbtionsladen weil die Spannung ja nur dann einbricht wenn ein Gerät Strom zieht..

#### Wenn man nun die Batterie aus Effizienzgründen weglassen möchte würde ich den Victron als störfaktor ausbauen!
Dieser Aufbau wäre dann:
_PV (auf passende Spannung verkabelt!) -> SUN WR 20-60V auf 230V AC_

#### Wenn das nicht einfach machbar ist macht es keinen Sinn überhaupt umzubauen denn das bisschen Effizienz ist den Aufwand nicht wert..

harambe88, 2024-12-11:
Vielen Dank für die ausführliche Erklärung. Jetzt ist mir klar, wo das Problem liegt.

Ich möchte die Batterie allerdings nicht generell abhängen, sondern nur dann füllen, wenn Leistung übrig ist. Das heißt es soll alles in den WR gehen und erst dann in die Batterie. Damit könnte man immer einen Mindestvorrat an Energie vorhalten.

Was ich auch verhindern will, ist dass die Batterie komplett leergesaugt wird, wenn ein größerer Verbraucher eingeschaltet wird. Das führt nämlich momentan dazu, dass ich (in dieser Jahreszeit) oft eine ganz leere Batterie habe. Ich habe das mit VBAT CUTOFF und VBAT  RESTART so eingestellt, dass die Batterie bis 25% aufgeladen wird und dann wieder bis 10% entladen wird.  Das geht dann den ganzen Tag rauf und runter. Im Sommer ist es natürlich alles anders. 

Wenn die Verluste durch Batterie laden/entladen gering wären (10-15%) hätte ich damit auch kein Problem. Die Verluste sind aber bei mir 35 %. Zumindest nach meiner Rechnung. Ich vergleiche da einfach eingespeiste zu entnommener Energie. Da kann eigentlich kein Rechenfehler drin sein. Vielleicht hat die Batterie auch einen Schaden. Aber das wäre ein ganz neues Thema.

erxbout, 2024-12-11:
Es könnte sein, dass der SUN sich einiges von der Effizienz genehmigt wenn er eine niedrige Standby Leistung erzeugt..

Der Victron (zumindest bei mir) lügt auch ein bisschen was den Ertrag angeht..
Zumindest so kann ich mir bei meinem System erklären, dass 2A richtung Batterie gehen aber der Smart Shunt nur 1,5A misst..

Aber auch nicht schlimm wenn genug Strom da ist..

Aaber wenn ichs richtig verstehe ist der Plan, dass bei 25% eine Art Bypass stattfinden soll?

Dann würde ich irgendwo eine Automatisiterung machen die den AC Setpoint OVR per mqtt setzt und zwar auf den bestehenden AC Setpoint + Ladeleistung in die Batterie - 10W oder so.. (max aber benötigte Leistung im Haushalt)

Dann bleibt die Hardware wie sie ist und 25% sind sicher..

Die cutoff voltage würde ich niedriger setzen oder die restprozent höher damit sich das nicht gegenseitig aushebelt..

harambe88, 2024-12-12:
Gute Idee mit der Automatisierung per MQTT - probiere ich aus.