trucki-eu/Trucki2Shelly-Gateway #58: Trucki Stick p-Error

marcel7162, 2025-02-15:
I have an T2SG Stick and after some Minutes the Stick Shows P-Error.

trucki-eu, 2025-02-15:
Hallo Marcel,
mit welcher Firmware-Version? 
Kannst Du mal die Ausgabe von http://<<T2SGIP>>/debug posten?
Danke
Gruß,
Trucki

marcel7162, 2025-02-15:
> Kannst Du mal die Ausgabe von http://<>/debug posten?

Habe das sowohl auf 1.14 und 1.13 bekommen. Schicke ich in 5 Minuten da ich
den Stick grade neu gestartet hsbe

marcel7162, 2025-02-15:
Jetzt läuft er seit einer Stunde. Gestern 3x den Fehler bekommen

trucki-eu, 2025-02-15:
Hallo Marcel,
Ohne log und debug output weiß ich leider auch nicht weiter. 
Gruß,
Trucki

marcel7162, 2025-02-15:
Debug
Modbus mode: SUN
Idf version: v4.4.4
pcb version: 1111
getFreeHeap(): 175128
getSdkVersion(): v4.4.4
getCpuFreqMHz(): 160
WiFi RSSI(dbm): -68
WiFi SSID: Smart home Internet
WiFi BSSID: 7C:FF:4D:CC:2B:45
WiFi localIP: 192.168.1.180
WiFi gatewayIP: 192.168.1.1
WiFi subNetMask: 255.255.255.0
WiFi primaryDNS: 192.168.1.1
NTP Server: pool.ntp.org[141.144.246.224] :
Date/Time: 15-02-2025 08:38:28 (Day)
Websrv jsononce (max. 2.700kb): 1604
Websrv jsonlive (max. 2.700kb): 1743
Meter json size (max. 3.096kb): 1730
Meter error string:
Readout Max Websrv poll: 0[ms]
Readout Max Websrv page: 5[ms]
Readout Max Websrv json once: 0[ms]
Readout Max Websrv json live: 0[ms]
Readout Max Meter : 99[ms]
Readout Max Meter Err : 0
Readout Max MQTT : 39[ms]
Readout Max UDP poll : 1[ms]
Readout Max SUN2 Err : 0
Readout Max SUN3 Err : 0
Readout Max Timer upd. : 0[ms]
Readout Max ZEPC upd. : 1[ms]
Readout Max Sun : 351[ms]
Readout Max MyWifi : 1[ms]
Readout Max Main : 450[ms]

marcel7162, 2025-02-15:
So sieht das aus.

marcel7162, 2025-02-15:
Hilft das weiter?

trucki-eu, 2025-02-15:
Leider nein. Das sieht normal aus. Vielleicht auch weil der Fehler nicht mehr auftritt.

marcel7162, 2025-02-15:
Der Fehler tritt immer noch auf. Nach einer Stunde.  Sun(1) lokal P-Error