trucki-eu/Trucki2Shelly-Gateway #59: SUN1000G2 steuern mit ACSETPOINTOVR über mqtt

best57, 2025-03-06:
Hallo,
mein SUN1000G2 hat das T2SG Board eingebaut mit Firmware 1.14 und mit ext. Antenne.
Es passiert alle 2-3 Tage, dass das T2SG nicht mehr auf die geänderten Wert im ACSETPOINTOVR reagiert. Es ist dabei egal, ob der ZEPC Wert auf 0, 1, 2 oder 3 steht. Das einzige was ich machen kann, ist ein Reboot des T2SG. Dann geht es wieder ein paar Tage.
Bin für jeden Hinweis dankbar.
SG, Stefan

trucki-eu, 2025-03-06:
Hallo Stefan,
der Fehler ist so zumindest nicht bekannt. Kommen denn im Fehlerfall die Werte noch im Webinterface unter MQTT an? Hast Du im Fehlerfall mal probiert mit dem MQTT Explorer Daten zu schicken? Hast Du mal probiert irgendeinen Wert an MeterOVR zu schicken? Dadurch müsste ZEPC automatisch auf 1 springen.

Gruß,
Trucki

best57, 2025-03-06:
Hallo Trucki,
 Ich verwende den ioBroker mit dem MQTT Adapter für die MQTT-Kommunikation. Die Werte von ACSETPOINTOVR kommen nicht im Webinterface an. Allerding funktioniert im Fehlerfall das senden von Werten über METEROVR schon. Das T2SG reagiert darauf auch prompt. Auch kann ich mit ZEPCOVR den Wert von ZEPC direkt auf 0, 1 oder 2 setzten. Beim nächten Fehlerfall werde ich den Test mit dem MQTT Explorer machen.
Gruß, Stefan