trucki-eu/Trucki2Shelly-Gateway #60: Steuerung Sun1 (local), Sun2 (Remote) und Sun3 (Remote)

Gunter66, 2025-04-12:
Hallo.

Kurze Frage zur Steuerung der SUN2 &3 durch den SUN1:
Auf welchen Wert beziehen sich die Prozente von **Turn ON at:** und **Turn OFF at**: ?

Gruß

G

alexbenisch, 2025-05-08:
Hi, vielen Dank für die Arbeit am trucki Stick.

Gibt es irgendwo Infos zum betreiben mehrerer lumentree 800 aus einem Speicher oder generell mit 2 remote Wechselrichtern von lumentree?

Ich wohne in einem Mehrfamilienhaus mit 2kwp PV Anlage und  einem 10kwh Speicher (2x 5kwh Felicity ESS Parallel)  der mit einem Victron MPPT 150/45 geladen wird. Kann man z.B. rein von der technischen Umsetzbarkeit her, insgesamt 3 lumentree 800 an je einem Haushalt/Apartment mit eigenem Stromzähler betreiben, wenn die lumentree 800 alle am gleichen Speicher hängen? Die Anlage ist so platziert, dass die alle drei Apartments bzw. deren Wohnungsnetz über ein Kabel am jeweiligen lumentree 800 angeschlossen werden könnten.

Der Grund für meine Frage ist, dass ich nicht weiß ob es technisch unbedenklich ist wenn die Netze der 3 lumentree 800 alle an dem einen Speicher betrieben werden. 

Danke 
alexbenisch

Gunter66, 2025-05-08:
Dürfte kein Problem sein. Ich habe sogar drei Lumentree an einer Batterie, die kaskadieren geregelt sind.
Wenn die eingestellten Voltwerte für VBat cutoff und VBat reboot bei allen gleich sind.

theunknownstarman, 2025-05-08:
Warum DREI Lumentree 800 und nicht EIN Lumentree 2000?

alexbenisch, 2025-05-08:
Danke für eure Hinweise.

Der entscheidende Punkt den ich zu klären versuche ist, 
ob es ein Problem ist wenn die Wechselrichter in 3 unterschiedlichen Wohnungen per Nulleinspeisung einspeisen.
Mir ist klar das es eine komische Frage ist, aber die Busbar an der die Wechselrichter und der Speicher hängen verbindet ja irgendwie dann die drei Wohnungen, auch wenn die drei Wohnungen einen gemeinsamen Hausanschluss haben.
Wahrscheinlich ist  das eine etwas esoterische Frage, ich wollte es nur abklären.