trucki-eu/Trucki2Shelly-Gateway #61: DAYENERGY / Day energy - zählt falsch hoch

HrzM83, 2025-04-13:
Hi,

geht um einen T2SG v 1.14 an einem SUN-600G2-M. Vor kurzem im ask2IT Shop gekauft.

DAYENERGY bzw Day energy sollte doch eingetlich die erzeugte Energie zeigen. Allerdings zählt er hoch ohne das Strom erzeugt wird.
Pünktlich um 00:00 wird der Wert auf 0 gesetzt, dann geht es in 0.1 kWh Schritten hoch, scheinbar alle 4 Stunde und 8 Minuten.

Der Wert direkt am Wechselrichter im Display ist korrekt auf  0.0

00:00 Uhr --> 0
04:08 Uhr --> 0.1
08:16 Uhr --> 0.2
12:24 Uhr --> 0.3
16:35 Uhr --> 0.4

Wenn Strom erzeugt wird ist der Wert über den Tag auch nicht korrekt. Gestern zeigt der T2SG 2.2 kWh im Display stand 1.9 kWh und ein Shelly der in der Leitung hängt stand auf 2.0 kWh.

Gruß Uwe

buddystk, 2025-09-18:
Hi,

das kann ich so auch bestätigen, gibt es evtl. schon einen Workaround oder eine Lösung?

Gruß
Stefan

trucki-eu, 2025-09-19:
Der T2SG-Stick ließt einfach nur den Wert aus dem SUN aus. Da kann ich von T2SG Seite her leider wenig tun.

HrzM83, 2025-09-19:
Wo die Daten herkommen kann ich leider auch nicht sagen. Aber im T2SG steht was andere als am Gerät(Display). 

Im Grafan-Screenshot ist zu sehen wie immer ein paar Wh hinzukommen. Im gezeigten Beispiel hat der Lumtree zischen 0 und ca. 13:00 Uhr nicht gewechselrichtet, trotzdem gehen die kWh hoch... 

[image]
[image]
[image]

trucki-eu, 2025-09-20:
Interessant das "Total engery" auch so auseinander läuft. Vielen Dank für die Bilder. Ich versuche es mal nachzustellen und schaue es mir mal an. 

Was ist das denn für eine Hardware? Lumentree feat. Trucki 2024 1000W, T2SG Stick ext. Antenne?

Gruß, Trucki

buddystk, 2025-09-20:
Hallo,

ich habe einen LumenTree feat. Trucki‍‍‍‍‍ SUN-800G2-M Modell 2024 und einen T2SG ohne ext. Antenne. FW 1.4

"Total engery" läuft bei mir auch auseinander. Heute wurde bei mir über Nacht bis ca. 04:30 Uhr eingespeist, hier scheint es so, dass die Werte übereinstimmen und um 8 Uhr gab es auch keinen 0,1 kWh Sprung. Müsste ich später mal schauen, ob der 0,1 kWh Sprung immer erst auftritt, wenn der LumenTree für mehr als 4 Stunden nicht einspeist und der T2SG sich im vBAT low Modus befindet.

HrzM83, 2025-09-20:
LumenTree feat. Trucki 
SUN-600G2-M incl. T2SG 
Modell 2024 
Trucki-kompatibel 
24V bis 48V / 600W x 1 

Ohne Antenne. FW 1.4