trucki-eu/Trucki2Shelly-Gateway #62: Shelly hat json Format geändert Shelly Pro 3em

donaldstod, 2025-05-22:
Hallo zusammen,
Shelly hat mit dem letzten SW Update 1.6.2 offensichtlich die json-Ausgabe geändert.
Seit dem Update ist ein auslesen des Shelly Pro 3em mit em:0,total_act_power und der gewohnten Konfiguration nicht mehr möglich.
Die Aussage von Shelly lautet man solle das json als object und nicht als string auslesen.
Zudem kam die Aussage, dass Shelly nicht zuständig sei für eine falsche Implementierung von Drittanbietern... 
@trucki-eu : Ich kann Deinen Code leider nicht einsehen, aber offensichtlich müsste man hier etwas an der Einleseart ändern um das T2SG auch zukünftig kompatibel mit den Shelly(s) zu machen.
Ich werde fürs erste keine Updates der Shellys mehr durchführen.
Viele Grüße 
Michael

erxbout, 2025-05-22:
Es wird das json eingelesen und mit dem Pfad "em:0,total_act_power" die Leistung rausgeholt..

Mit einem Beispiel json welches jetzt rauskommt, kann man diesen Pfad anpassen (auch ohne Firmware update)
Später kann das eingepflegt werden und damit würde es nach einem Firmware update wieder out of the box funktionieren..

Mit einem Beispiel vom neuen JSON kann ich den neuen selector herausfinden

side note: gut, dass ich noch keine Zeit hatte zum updaten xD

donaldstod, 2025-05-22:
Hi erxbout, 
danke für Deine schnelle Antwort :-)
Einen großen Unterschied konnte ich noch nicht feststellen, eigentlich sollte es noch funktionieren...dennoch geht es seit v1.6.2 nicht mehr :-( 
Danke für deine Unterstützung!
Hier der Output:
{"ble":{},"bthome":{},"cloud":{"connected":true},"em:0":{"id":0,"a_current":1.221,"a_voltage":231.1,"a_act_power":83.1,"a_aprt_power":281.8,"a_pf":0.29,"a_freq":50.0,"b_current":0.225,"b_voltage":230.9,"b_act_power":0.0,"b_aprt_power":51.8,"b_pf":0.00,"b_freq":50.0,"c_current":0.266,"c_voltage":231.7,"c_act_power":20.2,"c_aprt_power":61.6,"c_pf":0.33,"c_freq":50.0,"n_current":null,"total_current":1.712,"total_act_power":103.308,"total_aprt_power":395.264, "user_calibrated_phase":[]},"emdata:0":{"id":0,"a_total_act_energy":1727.01,"a_total_act_ret_energy":94.18,"b_total_act_energy":1477.93,"b_total_act_ret_energy":0.00,"c_total_act_energy":1816.82,"c_total_act_ret_energy":0.00,"total_act":5021.75, "total_act_ret":94.18},"eth":{"ip":null,"ip6":null},"modbus":{},"mqtt":{"connected":true},"sys":{"mac":"0CB815FCE598","restart_required":false,"time":"13:29","unixtime":1747913350,"last_sync_ts":1747911579,"uptime":18215,"ram_size":262064,"ram_free":79416,"ram_min_free":61888,"fs_size":524288,"fs_free":184320,"cfg_rev":40,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"btrelay_rev":0,"available_updates":{},"reset_reason":3,"utc_offset":7200},"temperature:0":{"id": 0,"tC":47.6, "tF":117.6},"wifi":{"sta_ip":"192.168.178.71","status":"got ip","ssid":"WLAN-001","rssi":-82,"ap_client_count":0,"sta_ip6":["fe80::eb8:15ff:fefc:e598","2003:ca:a70b:1e00:eb8:15ff:fefc:e598","fd21:a2b5:b60:0:eb8:15ff:fefc:e598"]},"ws":{"connected":false}}

scandrive, 2025-05-22:
Ich habe gestern meinen pro3 em auf 1.6.2 geupdatet und kann das nicht nachvollziehen.

T2SG v. 1.14

donaldstod, 2025-05-22:
Ich hab noch die T2SG v 1.1.2 drauf, boah danke.
Das könnte es natürlich sein, ich mach beim T2SG ein update auf v 1.1.4 und schreibe ob es geklappt hat.
Danke :-)

donaldstod, 2025-05-22:
Okay, ich nutze einen Wemos D1 Pro als Controller.
Für die Version 1.1.4 wird aber ein Esp8285 verwendet...
Der Wemos D1 Pro hat einen ESP8266, deshalb konnte ich die Version 1.1.4 nicht finden...
Im Shelly Forum konnte ich noch einen weiteren User finden der scheinbar das gleiche Problem hat.
https://shelly-forum.com/thread/29923-shelly-pro-3em-mit-trucky-t2sg871f21/?postID=307320#post307320

donaldstod, 2025-05-22:
Habe es wieder hinbekommen 🥳
Man muss http://192.168.178.71/rpc/EM.GetStatus?id=0
(IP Adresse natürlich anpassen 😉) verwenden und nur 
total_act_power
eintragen.
Dann kommt wieder nur ein String der ausgelesen werden kann.
Vielen lieben Dank für Eure Hilfe!

closed 2025-05-22 by donaldstod